tier4/scenario_simulator_v2 #543: Scenario Simulator 2 waits for Autoware even with launch_autoware:=False

prybicki, 2021-10-04:
**Describe the bug**
Scenario Simulator 2 waits for Autoware even with launch_autoware:=False. 

**To Reproduce**
Steps to reproduce the behavior:
Build Scenario Simulator and run 
```
ros2 launch scenario_test_runner scenario_test_runner.launch.py record:=False launch_autoware:=False scenario:=src/scenario_simulator_v2/external/operational-design-domains/cargo_delivery/scenarios/UC-001-0010-Kashiwa.yaml 
```
Error:
```
[openscenario_interpreter_node-3] [INFO] [1633364873.961957193] [simulation.openscenario_interpreter]: AutowareError: Simulator waited 30 seconds, expecting the Autoware state to transitioning to InitializingVehicle, but there was no change. The current Autoware state is NOT PUBLISHED YET.
```

**Expected behavior**
Scenario Simulator 2 should run without waiting for Autoware and simulate NPC positions (e.g. send updates over ZMQ).

**Desktop (please complete the following information):**
 - OS: Linux Mint 20.2 (aka Ubuntu 20.04)
 - ROS2 galactic
 - DDS cyclonedds

hakuturu583, 2022-03-07:
@prybicki Thank you!

hakuturu583, 2023-07-12:
This issue is no longer active and features proposed in this issue was not merged in #705 
So, I close this issue.
Please re-open this issue if necessary.
Thank you for sending comment @prybicki .

closed 2023-07-12 by hakuturu583