tier4/scenario_simulator_v2 #601: Error in scenario conversion (YAML->XOSC) documentation or an absent executable

khaemn, 2021-11-04:
**Describe the bug**
It is impossible to convert a TierIV YAML scenario to the OpenScenario as described in the online documentation

**To Reproduce**
Ubuntu 20.04
Steps to reproduce the behavior:
0. Install ROS2 Foxy, clone and build the `scenario_simulator_v2` repo, activate all the necessary environments
1. Try convert an `.yaml` scenario file to an `.xosc` according to the https://tier4.github.io/scenario_simulator_v2-docs/user_guide/scenario_test_runner/ScenarioFormatConversion/
2. No such executable found, no even a file named `scenario_test_utility.py` can be found in the repository

**Expected behavior**
The `.yaml` file is converted to an `.xosc` file

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Version: master branch commit 2a79b63d 2021-10-11
 - ROS2 Foxy

YoshinoriTsutake, 2021-11-10:
https://tier4.github.io/scenario_simulator_v2-docs/user_guide/scenario_test_runner/ScenarioFormatConversion/
The instruction above is too old. 
So, please use the command below instead, 
`ros2 run openscenario_utility yaml2xosc --input </PATH/TO/SCENARIO.yaml> --output <DIRECTORY>`

Also, a PR to fix this issue is supposed to be open soon, and then the instruction is updated.

yamacir-kit, 2021-11-12:
> Also, a PR to fix this issue is supposed to be open soon, and then the instruction is updated.
#611 fixed this issue.

closed 2021-11-12 by yamacir-kit