tier4/scenario_simulator_v2 #700: Parameter 'timeout' has invalid type Error  in doccs/tutorials/RunWithDocker.md

xczhanjun, 2022-02-12:
**Parameter 'timeout' has invalid type Error  in doccs/tutorials/RunWithDocker.md**

[traffic_simulation_demo-1] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[traffic_simulation_demo-1]   what():  parameter 'timeout' has invalid type: Wrong parameter type, parameter {timeout} is of type {double}, setting it to {integer} is not allowed.

**To Reproduce**
Steps to reproduce the behavior:
 rocker --x11 tier4/scenario_simulator_v2:galactic ros2 launch cpp_mock_scenarios mock_test.launch.py scenario:=crashing_npc scenario:=traffic_simulation_demo launch_rviz:=true timeout:=60

hakuturu583, 2022-03-04:
Thank you for checking our mistake!
I will fix soon!

hakuturu583, 2022-04-01:
This problem was solved by #715

closed 2022-04-01 by hakuturu583