tier4/scenario_simulator_v2 #765: Fail to run test scenarios in docker

shmpwk, 2022-04-20:
**Describe the bug**
In the tutorial, [Run test scenarios in docker](https://tier4.github.io/scenario_simulator_v2-docs/tutorials/RunTestScenarioWithDocker/#run-test-scenarios-in-docker) does not work.

First of all,  command ` $ rocker --x11 --oyr-mount $PWD/scenarios/UC-001-0001-Kashiwa.yaml -- tier4/scenario_simulator_v2 ros2 launch scenario_test_runner scenario_test_runner.launch.py`  is not valid now. 
Instead, command `$ rocker --x11 --oyr-mount $PWD/scenarios/UC-001-0001-Kashiwa.yaml -- tier4/scenario_simulator_v2:galactic ros2 launch scenario_test_runner scenario_test_runner.launch.py`

However, it fails to launch and  the error says that the lifetimecycle is unconfigured.
```
[scenario_test_runner-1] [ERROR] [1650413557.603908415] [simulation.scenario_test_runner]: Failed to configure interpreter
[scenario_test_runner-1] [ERROR] [1650413557.604781543] [simulation.scenario_test_runner]: Interpreter is unconfigured now, but inactive expected.
```

I check UC-001-0001-Kashiwa.yaml and it executes `$(find-pkg-share cargo_delivery)`
but I cannot found `cargo_delivery` 
```
$ ros2 pkg prefix cargo_delivery
Package not found
```
So it seems we have to install `cargo_delivery` in advance? or the yaml is out of date?

The whole output is as follows:
```
$ rocker --x11 --oyr-mount $PWD/scenarios/UC-001-0001-Kashiwa.yaml -- tier4/scenario_simulator_v2:galactic ros2 launch scenario_test_runner scenario_test_runner.launch.py scenario:=/home/shumpeiwakabayashi/scenario_simulator_ws/src/scenario_simulator_v2/scenarios/UC-001-0001-Kashiwa.yaml launch_rviz:=True
Extension oyr_cap_add doesn't support default arguments. Please extend it.
Extension oyr_cap_drop doesn't support default arguments. Please extend it.
Extension oyr_colcon doesn't support default arguments. Please extend it.
Extension oyr_mount doesn't support default arguments. Please extend it.
Extension oyr_run_arg doesn't support default arguments. Please extend it.
Extension oyr_spacenav doesn't support default arguments. Please extend it.
Extension volume doesn't support default arguments. Please extend it.
Active extensions ['oyr_mount', 'x11']
Writing dockerfile to /tmp/tmpc41ljbfk/Dockerfile
vvvvvv
# Preamble from extension [oyr_mount]

# Preamble from extension [x11]

FROM tier4/scenario_simulator_v2:galactic
USER root
# Snippet from extension [oyr_mount]

# Snippet from extension [x11]

^^^^^^
Building docker file with arguments:  {'path': '/tmp/tmpc41ljbfk', 'rm': True, 'nocache': False, 'pull': False}
building > Step 1/2 : FROM tier4/scenario_simulator_v2:galactic
building >  ---> 5959b93f732d
building > Step 2/2 : USER root
building >  ---> Using cache
building >  ---> befb4b2cfefc
building > Successfully built befb4b2cfefc
Executing command: 
docker run --rm -it -v /home/shumpeiwakabayashi/scenario_simulator_ws/src/scenario_simulator_v2/scenarios/UC-001-0001-Kashiwa.yaml:/home/shumpeiwakabayashi/scenario_simulator_ws/src/scenario_simulator_v2/scenarios/UC-001-0001-Kashiwa.yaml  -e DISPLAY -e TERM   -e QT_X11_NO_MITSHM=1   -e XAUTHORITY=/tmp/.docker81mxnvmp.xauth -v /tmp/.docker81mxnvmp.xauth:/tmp/.docker81mxnvmp.xauth   -v /tmp/.X11-unix:/tmp/.X11-unix   -v /etc/localtime:/etc/localtime:ro  befb4b2cfefc ros2 launch scenario_test_runner scenario_test_runner.launch.py scenario:=/home/shumpeiwakabayashi/scenario_simulator_ws/src/scenario_simulator_v2/scenarios/UC-001-0001-Kashiwa.yaml launch_rviz:=True
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-04-20-09-12-35-406001-6f5c18362c68-1
[INFO] [launch]: Default logging verbosity is set to INFO
architecture_type       := awf/universe
autoware_launch_file    := planning_simulator.launch.xml
autoware_launch_package := autoware_launch
global_frame_rate       := 30.0
global_real_time_factor := 1.0
global_timeout          := 180
initialize_duration     := 30
launch_autoware         := True
launch_rviz             := True
output_directory        := /tmp
port                    := 8080
record                  := True
scenario                := /home/shumpeiwakabayashi/scenario_simulator_ws/src/scenario_simulator_v2/scenarios/UC-001-0001-Kashiwa.yaml
sensor_model            := 
vehicle_model           := 
workflow                := /dev/null
sigterm_timeout         := 8
[INFO] [scenario_test_runner-1]: process started with pid [64]
[INFO] [simple_sensor_simulator_node-2]: process started with pid [66]
[INFO] [openscenario_interpreter_node-3]: process started with pid [68]
[INFO] [openscenario_visualization_node-4]: process started with pid [70]
[INFO] [rviz2-5]: process started with pid [72]
[scenario_test_runner-1] [INFO] [1650413556.600185803] [simulation.scenario_test_runner]: Run UC-001-0001-Kashiwa.xosc (1 of 1)
[scenario_test_runner-1] [ERROR] [1650413557.603908415] [simulation.scenario_test_runner]: Failed to configure interpreter
[scenario_test_runner-1] [ERROR] [1650413557.604781543] [simulation.scenario_test_runner]: Interpreter is unconfigured now, but inactive expected.
[scenario_test_runner-1] [INFO] [1650413557.605071644] [simulation.scenario_test_runner]: Shutdown interpreter.
[scenario_test_runner-1] /home/shumpeiwakabayashi/scenario_simulator_ws/src/scenario_simulator_v2/scenarios/UC-001-0001-Kashiwa.yaml
[INFO] [scenario_test_runner-1]: process has finished cleanly [pid 64]
[INFO] [launch]: process[scenario_test_runner-1] was required: shutting down launched system
[INFO] [rviz2-5]: sending signal 'SIGINT' to process[rviz2-5]
[INFO] [openscenario_visualization_node-4]: sending signal 'SIGINT' to process[openscenario_visualization_node-4]
[INFO] [openscenario_interpreter_node-3]: sending signal 'SIGINT' to process[openscenario_interpreter_node-3]
[INFO] [simple_sensor_simulator_node-2]: sending signal 'SIGINT' to process[simple_sensor_simulator_node-2]
[openscenario_visualization_node-4] [INFO] [1650413558.812376422] [rclcpp]: signal_handler(signal_value=2)
[openscenario_interpreter_node-3] [INFO] [1650413558.812806248] [rclcpp]: signal_handler(signal_value=2)
[simple_sensor_simulator_node-2] [INFO] [1650413558.813335045] [rclcpp]: signal_handler(signal_value=2)
[INFO] [openscenario_visualization_node-4]: process has finished cleanly [pid 70]
[INFO] [simple_sensor_simulator_node-2]: process has finished cleanly [pid 66]
[INFO] [openscenario_interpreter_node-3]: process has finished cleanly [pid 68]
[INFO] [rviz2-5]: process has finished cleanly [pid 72]
```

**To Reproduce**
Steps to reproduce the behavior:
1. Follow [Run test scenarios in docker](https://tier4.github.io/scenario_simulator_v2-docs/tutorials/RunTestScenarioWithDocker/#run-test-scenarios-in-docker) 

**Expected behavior**
Succeed to run test scenario and to show it in rviz.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Ubuntu20.04
 - ROS2 galactic

hakuturu583, 2022-04-20:
@shmpwk Oh... cargo delivery package was already removed and I forget to update this document.
Thank you for finding my mistake.

hakuturu583, 2022-04-25:
@shmpwk I think this problem was fixed by this PR.
https://github.com/tier4/scenario_simulator_v2/pull/769

closed 2022-04-28 by shmpwk