tier4/scenario_simulator_v2 #767: NPC doesn't stop at the defined destination point in scenario yaml file and passes through it

mehmetdogru, 2022-04-22:
**Describe the bug**

NPC doesn't stop at the defined destination point in scenario yaml file and passes through it. In both cases with or without launching autoware along with the scenario_test_runner this issue happens. So for the sake of decreasing the complexity of the issue, without autoware launch option will be given as the step to reproduce the issue.

**To Reproduce**

1. Basically create any scenario using any lanelet2_map with [tierIV's scenario editor](https://scenario.tier4.jp/scenario_editor) which includes an NPC with a given destination point or [download issue_sample_data](https://drive.google.com/file/d/1N_W5L1dumQR2Mb5tUwCSHfoC9iJhdEXK/view?usp=sharing).
2. Be aware of the destination point of NPC. If issue_sample_data is downloaded import .yaml file to [tierIV's scenario editor](https://scenario.tier4.jp/scenario_editor) to see where is the destination point for NPC. 
3. Launch scenario_test_runner with the command given below:
`ros2 launch scenario_test_runner scenario_test_runner.launch.py sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle scenario:=<path-to-yaml-file> launch_rviz:=true launch_autoware:=false`
4. Observe the issue.

Note: If you want to try launching autoware.universe along with scenario_test_runner to reproduce the issue and if you downloaded issue_sample_data then you must change isEgo property with 'true' in the yaml file.

**Expected behavior**

NPC should stop at defined destination point.

**Screenshots**

[image]
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Browser: Chrome Version 100.0.4896.60  
 - ROS2 version: galactic
 - DDS: rmw_cyclonedds_cpp

yamacir-kit, 2022-04-23:
@mehmetdogru 

**This is a known and intended behavior.** The reasons for this are explained below.

The scenarios created by the TIER IV scenario editor are output as ASAM OpenSCENARIO 1.x-based YAML scenario files. Here, the destination of the entity specified in the scenario editor corresponds to the AcquirePositionAction in OpenSCENARIO 1.x.

Since Autoware stops when the destination is reached, it would seem that the desired behavior would be for the NPC to also automatically stop when the destination is reached. At the same time, however, we know from experience that there are cases where it is more convenient for the NPC to continue driving at a speed and route that generally seems natural, rather than stopping after reaching its destination.

Unfortunately, OpenSCENARIO does not specify the subsequent behavior of an entity that has reached its destination via an AcquirePositionAction. In other words, according to the OpenSCENARIO standard, it does not matter whether the NPC stops at the destination or not. Therefore, this behavior is not a bug.

Therefore, we require the user to explicitly describe in the scenario if he wants the NPC to stop at the destination. That is, you must use the ReachPositionCondition to determine that the destination has been reached in the AcquirePositionAction and then indicate zero speed by the SpeedAction. Using this method, you can write highly interchangeable scenarios that will work as expected, whether the simulator stops the NPC at the destination or not.

yamacir-kit, 2022-04-26:
There is currently no way to set the upper speed limit for NPCs other than the upper speed limit embedded in the Lanelet.

This is not unintentional, we simply forgot to implement it. So I will add this issue to our backlog.

The upper speed limit cannot be set, but SpeedAction can be used to maintain the specified speed. If what you are looking for is not to make the upper speed limit lower than the value embedded in the lanelet, but to make it run at a specified speed, then using SpeedAction is the simplest solution.

mehmetdogru, 2022-04-26:
@yamacir-kit 

Actually all the bus odd scenarios created by odd-wg uses the same map. And in different scenarios there are different NPCs moving at different speeds varies from 10 to 50 kph. In this map _speed_limit_ should be 50 kph so that all the scenarios can be run using the same lanelet2_map.

I believe it is rather an important issue to track also for me so if you don't mind I will be creating an issue in this repository as well. Thank you!

closed 2022-04-26 by mehmetdogru