tier4/scenario_simulator_v2 #770: Is it possible to set the paths for osm/pcd files relative from specified yaml file ?

soblin, 2022-04-26:
Hi,

When I specify the path of yaml file in launching scenario_simulator, like 

```
ros2 launch scenario_test_runner scenario_test_runner.launch.py scenario:='/home/<user>/Downloads/t4v2.yaml'
```

it searches for pcd/osm file written in that yaml file. So we have to write absolute paths for them in that yaml file as follows:

```yaml
  RoadNetwork:
    LogicFile:
      filepath: /home/<user>/Downloads/lanelet2_map.osm
    SceneGraphFile:
      filepath: /home/<user>/Downloads/lanelet2_map.pcd
```

but rewriting the absolute path every time after I downloaded the yaml file from [scenario editor](https://scenario.tier4.jp/scenario_editor/restricted/) is a bit cumbersome, so hopefully I'd like it to search for pcd/osm files using **relative path from the yaml file if the path does not start with** '/'. 

I had a quick look at [openscenario_interpreter.cpp](https://github.com/tier4/scenario_simulator_v2/blob/b67ff4bb1b3b15ceee5cf492b1a84b43a12770ae/openscenario/openscenario_interpreter/src/openscenario_interpreter.cpp#L69-L83) and found out the item ["RoadNetwork"]["Logicfile"]["filepath"] that we re-write appears in the variable `logic_file`. So I think checking if `logic_file` starts with '/' and converting it to absolute path otherwise will work out like in the following.

```
const auto logic_file = currentScenarioDefinition()->road_network.logic_file;
+ if (not logic_file.string()[0] == '/') {
+    logic_file = <dir of yaml file>/logic_file;
+ }
auto configuration = traffic_simulator::Configuration(
  logic_file.isDirectory() ? logic_file : logic_file.filepath.parent_path());
```

In summary if logic_file is "lanelet2_map.pcd" , "./lanelet2_map.pcd", or whatever relpath from the yaml file that does not start with '/', we convert it to absolute path.

How do you think about this feature/approach ? thanks

yamacir-kit, 2022-04-26:
@soblin 

**Currently, relative paths are intentionally not supported.**

First, since scenario_simulator_v2 is created as a ROS2 node, the relative path from the simulator in a simple implementation starts in the install directory of the ROS2 workspace. This specification was not adopted because it would result in unintended behavior for non-ROS2-friendly users.

The next most likely solution is to use the directory where `ros2 launch scenario_test_runner scenario_test_runner ...` was executed (the current working directory) as the starting point. This seems more natural than the previous method, but on the other hand, it was not adopted because it would result in unintended behavior for ROS2 users.

Accepting relative paths starting from the YAML file of the scenario you are proposing is another good idea.

However, we did not adopt either of these because if we imposed certain rules on the arrangement of map files for the simulator, it might affect the arrangement of map directories in Autoware itself in the future.
The organization of Autoware's map files may change in the future. For example, if Autoware needs to autonomously drive over a larger area than will fit in a single osm, the design may change so that multiple map files are loaded sequentially instead of one.
If scenario_simulator_v2 had its own map file placement rules, this would limit the scope of Autoware's architectural design in terms of compatibility with previously written scenarios.
We do not want the rules imposed by scenario_simulator_v2 to be a drag on the Autoware design.

Therefore, we only support absolute paths, which is the only reliable and stable way.

We recognize that this is an issue that needs to be addressed, as we often get comments similar to yours from within the company. However, we do not believe it is the time to do so, as it would give us maximum freedom in the design of Autoware's architecture.

hakuturu583, 2023-07-12:
I think this issue was no longer active.
Please re-open issue if necessary.

hakuturu583, 2023-07-12:
Thank you for sending your idea. @soblin

closed 2023-07-12 by hakuturu583