tier4/scenario_simulator_v2 #775: NPCs should be able to get max speed parameter from scenario yaml file

mehmetdogru, 2022-04-27:
This issue is a part of https://github.com/autowarefoundation/autoware.universe/issues/566

**Description**
NPCs currently take max speed from related lanelet2_map's speed limit instead of scenario yaml file. And when there is multiple scenarios using the same lanelet2_map with different NPC speed configurations which also include NPCs moving lower speeds than lanelet2_map's speed limit, the feature proposed here is becoming a necessity.

**Describe the solution you'd like**
NPC vehicles should be able to take the argument for maximum speed that the can move from yaml file and if it is lower than embedded speed limit on the lanelet2_map, the parameter from yaml file must be accepted as max speed.

max_NPC_speed = _min_(`embedded_speed_limit_on_lanelet2` , `max_speed_from_yaml_file`)

mehmetdogru, 2022-05-22:
Setting `Initial speed` instead of `Speed` parameter makes the difference. Even though it is written `Initial speed` in the scenario editor it is interpreted as `AbsoluteTargetSpeed` so NPCs take that as a target speed.

mitsudome-r, 2022-06-06:
I've also confirmed this. It seems there is something wrong with treating `Initial speed` and `speed` (target speed) parameter either in the scenario editor or scenario runner.

* Sample scenario: 
[UC-PB-001-0003-1-Hsinchu.yaml.txt](https://github.com/tier4/scenario_simulator_v2/files/8842491/UC-PB-001-0003-1-Hsinchu.yaml.txt) (modify map file path and remove `.txt` from the file name) 
* Map: https://gitlab.com/autowarefoundation/operational-design-domains/-/tree/master/public_road_bus/maps/hsinchu

In this scenario, intial speed is set 0, and target speed is set to 2.8m/s (10km/h) so it should gradually speed up to 10km/h, but never starts moving when running the scenario. ([video link](https://youtu.be/w1Vje0iHMTU))
[image]

hakuturu583, 2024-02-14:
@mehmetdogru Is it possible to set a maximum speed for NPCs?
This Issue is no longer considered Active, so if I don't hear back from you in a week, I will Close it.

mehmetdogru, 2024-02-15:
@hakuturu583 

Thank you for reminder.

Well, it is kind of possible by setting `Initial speed`[_1_] but not quite. Although it is not actually initial speed it is set but `AbsoluteTargetSpeed`[_3_] and vehicle starts and drives with that speed only. Whereas `Speed`[_2_] does not have any affect. (corresponding to `maxSpeed`[_4_]) 

Meaning, as in the example case below, Npc1 will spawn at a speed `1 m/s` and drives at `1 m/s` regardless of `maxSpeed`[_4_]. So it does not matter whether `maxSpeed`[_4_] is `0.0 m/s` or `3.0 m/s`

As a conclusion, in reality we do not have initial or max speed but only a target speed (or speed). So IMO there are 2 ways to eliminate this confusion:

**Option 1)** 
- Use `Initial speed`[_1_] (e.g. `1m/s`) as initial speed in the scenario meaning NPC1 would start at that speed (`1m/s`) and speed up to `Speed`[_2_] (e.g. `3 m/s`).
- Rename `Speed`[_2_] as _Max Speed_

**Option 2)** 
- Remove `Speed`[_2_] unless it is used somewhere else.
- Rename `Initial speed`[_1_] as Absolute _Target Speed_ or _Speed_

I would prefer the **Option 1** :bow:

hakuturu583, 2024-02-16:
Checking the OpenSCENARIO notation, the initial speed of NPC1 is spawned at 1.0 m/s because SpeedAction is executed with `dynamics_shape=step`, and then the maximum speed is specified as 3.0 m/s by AssignControlelrAction. AssignControlelrAction.
To increase the speed to 3.0 m/s, it is necessary to update the target speed by executing SpeedAction again.

closed 2025-04-11 by hakuturu583