tier4/scenario_simulator_v2 #776: autoware.universe is not engaging when there is an iterative scenario

mehmetdogru, 2022-04-27:
**Description**
When an iterative parameter exists in scenario definition, after the first iteration second one starts. If the NPC mode is active for the _scenerio_test_runner_(_autoware.universe_ is not running) iterative scenarios is realizing and finishing after all the iterations. 

However if one runs _autoware.universe_ along with _scenerio_test_runner_, _autoware.universe_  is not engaging in the time specified as `initialize_duration` so _openscenario_interpreter_node_ is shutting down. So one cannot test the rest of the scenario.

Here is the related part after the second iteration starts:
```python
[openscenario_interpreter_node-3] [INFO] [1651075172.607175842] [simulation.concealer]: Simulator waiting for Autoware state to be Initializing (2).
[openscenario_interpreter_node-3] [INFO] [1651075173.607170647] [simulation.concealer]: Simulator waiting for Autoware state to be Initializing (1).
[openscenario_interpreter_node-3] [INFO] [1651075174.607185735] [simulation.concealer]: Simulator waiting for Autoware state to be Initializing (0).
[openscenario_interpreter_node-3] [INFO] [1651075175.581202385] [simulation.concealer]: [32mShutting down Autoware: (1/3) Stop publishing/subscribing.[0m
[openscenario_interpreter_node-3] [INFO] [1651075175.606130081] [simulation.concealer]: Autoware is DRIVING now.
[openscenario_interpreter_node-3] [INFO] [1651075175.606570827] [simulation.concealer]: Autoware is DRIVING now.
[openscenario_interpreter_node-3] [INFO] [1651075175.607958523] [simulation.concealer]: Autoware is DRIVING now.
[openscenario_interpreter_node-3] [INFO] [1651075175.608341434] [simulation.concealer]: Autoware is DRIVING now.
[openscenario_interpreter_node-3] [INFO] [1651075175.608605849] [simulation.concealer]: Autoware is DRIVING now.
[openscenario_interpreter_node-3] [INFO] [1651075175.836675413] [simulation.openscenario_interpreter]: [1;31mAutowareError: Autoware did not reach an engageable state within the specified time (initialize_duration).[0m
[openscenario_interpreter_node-3] [INFO] [1651075179.055190713] [rclcpp]: signal_handler(signal_value=2)
```
**Full Log:**
[scenario_iteration_issue.log](https://github.com/tier4/scenario_simulator_v2/files/8574704/scenario_iteration_issue.log)

**To Reproduce**

1. Download the [map data](https://gitlab.com/autowarefoundation/operational-design-domains/-/tree/master/public_road_bus/maps/hsinchu) and [the scenario file](https://gitlab.com/autowarefoundation/operational-design-domains/-/blob/master/public_road_bus/UC-PB-001-0004-1-Hsinchu.yaml).
2. [Run scenario_test_runner with autoware](https://github.com/autowarefoundation/autoware_launch/pull/26).
3. Observe that after the first iteration the other iterations won't be realized.

**Expected behavior**

All the iterations must be run automatically along with autoware.universe.

**Versions:**
- **scenario_simulator**:  7f5bf7e5b9269004e85adfb5cfc7495197c2208c
- **autoware.universe**: [`7ed3e9a`](https://github.com/autowarefoundation/autoware.universe/commit/7ed3e9aac77460639839d4e7f67d21725cde02b5)
- **OS**: Ubuntu 20.04
- **ROS2**: galactic
- **DDS**: rmw_cyclonedds_cpp

mitsudome-r, 2022-05-16:
This might be caused due to the old instruction from PR ( which launch autoware separately).
I just told @mehmetdogru to run scenario simulator differently and see the results first.

mitsudome-r, 2022-05-16:
He was able to run the second parameter fine with the new command so closing this issue.

closed 2022-05-16 by mitsudome-r

mehmetdogru, 2022-05-17:
Just for convenience, here is how to start `scenario_test_runner` without having the issue described with details [above](https://github.com/tier4/scenario_simulator_v2/issues/776#issue-1217627602):

```bash
source ~/autoware/install/setup.bash
ros2 launch scenario_test_runner scenario_test_runner.launch.py sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle scenario:=<path_to_yaml_file> architecture_type:=awf/universe launch_rviz:=true launch_autoware:=true
```
**!** Notice that `launch_autoware:=true`

**!** Make sure that you **don't launch** `planning_simulator` separately with another command.