tier4/scenario_simulator_v2 #781: Scenario Test Runner not working with autoware

adamkrawczyk, 2022-05-16:
**Describe the bug**
Using [pilot-auto](https://github.com/tier4/pilot-auto), commit id: `c92557c9698cd040da57b53c7cdde533128d9054` I'am unable to make autoware and scenario test runner works

**To Reproduce**
Steps to reproduce the behavior:
1. Clone https://github.com/tier4/pilot-auto
2. Build
3. run example by `ros2 launch scenario_test_runner scenario_test_runner.launch.py scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml' launch_rviz:=true record:=false vehicle_model:=lexus sensor_model:=aip_xx1`
4. See an error

**Expected behavior**
Ego driving

**Desktop:**
 - OS: Ubuntu
 - Version 20.04
 - ROS2 Galactic
 - Cyclone DDS

**Error log**

Autoware is initialized but in ` [simulation.concealer::waitForAutowareStateToBeWaitingForRoute] Simulator waiting for Autoware state to be WaitingForRoute (3).` state. In this state, it hung till timeout causing exit. 

NOTE: 
Interesting logs

```
Info: [planning.mission_planning.mission_planner::publishRoute] Route successfully planned. Publishing...
Info: [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner::waitForData] waiting for scenario topic
Error : [planning.scenario_planning.lane_driving.motion_planning.motion_planning_container::OnLoadNode] Failed to find class with the requested plugin name 'surround_obstacle_checker::SurroundObstacleCheckerNode' in the loaded library
```

```
Info:  [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner::waitForFirstPose] waiting for self pose...
Warning:  [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner::resetPlanning] [ObstacleAvoidancePlanner] Reset planning
Warning:  [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner::getLatestTransform] failed to get transform from map to base_link: "map" passed to lookupTransform argument target_frame does not exist. 
```

Full log: 
[log.txt](https://github.com/tier4/scenario_simulator_v2/files/8699160/log.txt)

I can also provide a rosbag if needed.

The initial pose seems to be set as you can see on the screen-shot
[image]

I wonder if there might be a problem with openscenario <-> t4s2.yaml since there is a log 

```
[openscenario_interpreter_node-3] [path_distance_calculator_node-61] [WARN] [1652696567.467424226] [autoware_api.utils.path_distance_calculator]: no path
```

yamacir-kit, 2022-05-16:
@adamkrawczyk 
Will this behavior be reproduced no matter how many times I rerun the simulator?
In the sample scenario, Autoware's initial location is slightly randomized, which can cause Autoware to fail in route planning.

adamkrawczyk, 2022-05-16:
@yamacir-kit 
Yes, this is no matter how many times I try, also extending initialization time does not work (tested 60s). 

Also does not work with other scenarios I have been using before.

yamacir-kit, 2022-05-16:
@adamkrawczyk 
Looking at the logs you provided, it seems that `rosbridge_server` failed to start. Did you refresh your terminal after the build and then source install/setup.bash?

adamkrawczyk, 2022-05-16:
@yamacir-kit 
Thank you!! It turned out that it was not installed - I have done rosdep step but supposedly `rosbridge_server` is not a part of any `package.xml`. I have also a problem building with `quaternion_operation` and added this to the package. 

Anyway thank you for your super fast support. It is working now. 
Should I rise an issue on for example autoware repo or will you manage this issue?

adamkrawczyk, 2022-05-16:
There it is. Sorry I have missed this.  In the future I will start by looking for `PackageNotFoundError`

```
[openscenario_interpreter_node-3] future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> ```
exception=PackageNotFoundError("package 'rosbridge_server' not found, searching: [blablabla]")>
[openscenario_interpreter_node-3] Traceback (most recent call last):
[openscenario_interpreter_node-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/ament_index_python/packages.py", line 50, in get_package_prefix
[openscenario_interpreter_node-3]     content, package_prefix = get_resource('packages', package_name)
[openscenario_interpreter_node-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/ament_index_python/resources.py", line 48, in get_resource
[openscenario_interpreter_node-3]     raise LookupError(
[openscenario_interpreter_node-3] LookupError: Could not find the resource 'rosbridge_server' of type 'packages'
[openscenario_interpreter_node-3] ```

yamacir-kit, 2022-05-17:
@adamkrawczyk 

https://github.com/tier4/pilot-auto/commit/95c47a2b59090e89fe4af0c0f3acd0e8e4168c88

Apparently, the previous commit of the pilot-auto version you used (currently tier4/universe HEAD) intentionally removed the dependency on rosbride.
I don't know why it was removed, but I recommend raising the issue in the pilot-auto repository as it would prevent pilot-auto from working with scenario_simulator_v2 without rosbridge_server.

adamkrawczyk, 2022-05-20:
Closed due to opening https://github.com/tier4/pilot-auto/issues/471

closed 2022-05-20 by adamkrawczyk