tier4/scenario_simulator_v2 #856: bug scenario_test_runner speed limit doesn't work

beyzanurkaya, 2022-08-15:
**Describe the bug**

I created several scenarios with the scenario editor. these scenarios had speed limit iterations. I tried to ensure that the speed of the vehicle is fixed at 10, 20 and 30 km / h. However, I could not achieve this. Even though I ran the same scenario without changing anything, the vehicle sometimes stayed true to the speed limit I set, but in some cases it did not. I tried the scenarios from [ODD Repository](https://gitlab.com/autowarefoundation/operational-design-domains/-/tree/master/public_road_bus) as I thought I might have created the scenarios that I created incorrectly. but still faced the same problems.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit this command according to you and run 
```
ros2 launch scenario_test_runner scenario_test_runner.launch.py sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle \
scenario:=path/your/scenario \
architecture_type:=awf/universe launch_rviz:=false launch_autoware:=true
```
2. Observe vehicle velocity.

**Expected behavior**

```
   ControllerAction:
     AssignControllerAction:
       Controller:
         name: ''
         Properties:
            Property:
             - name: maxSpeed
                value: $vel
```
The above section belongs to the file of the scenario. ` value: $vel `  represents the maximum speed of the EGO. EGO must stick to the specified speed limit.

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
- OS: Ubuntu20.04
- autoware.universe
- ROS2

And also I added a senario file. You can try this one. 

**I converted yaml file to text so I can upload.  Before run this scenario you have to change file extension to .yaml**

[UC-PB-001-0006-Hsinchu.txt](https://github.com/autowarefoundation/autoware.universe/files/9267043/UC-PB-001-0006-Hsinchu.txt)

beyzanurkaya, 2022-08-15:
setting speed limit works commit [970f517](https://github.com/tier4/scenario_simulator_v2/commit/970f5172929b9bd25e6d34103965a65753ee1245)

hakuturu583, 2022-08-16:
@beyzanurkaya 
> The above section belongs to the file of the scenario. value: $vel represents the maximum speed of the EGO. EGO must stick to the specified speed limit.

It seems, ego vehicle (autoware vehicle) state is waitnig for engage and does not started simulation yet.

We want to reproduce the situation just in case.
Can you give us some information about the version of Autoware?

beyzanurkaya, 2022-08-16:
@hakuturu583 

I use autoware.universe and I checked out the last main

piotr-zyskowski-rai, 2022-08-17:
@beyzanurkaya thanks for reporting the issue. 

This issue seems to be very similar to one I was working on and which I solved in [this draft PR](https://github.com/tier4/scenario_simulator_v2/pull/848). It solves the problem but requires some refactoring. Therefore, I will assign myself to this issue.

beyzanurkaya, 2022-08-30:
@piotr-zyskowski-rai Is there any update this issue?

hakuturu583, 2023-07-12:
@piotr-zyskowski-rai It seems, you claim that you solved this problem in #848 
Is it right?

piotr-zyskowski-rai, 2023-07-12:
@hakuturu583 @beyzanurkaya 
That is correct, #848 fixed the setting of the velocity limit. To make sure I tested it locally again on the scenario you provided. I took one from [this commit](https://gitlab.com/autowarefoundation/operational-design-domains/-/commit/8691188c143769bf6c9605bf8ff12dda64265104) since the Hsinchu map was deleted soon after from the repository. Ego vehicle's max speed was 30, 40, and 50km/h as expected in the scenario.

hakuturu583, 2023-07-14:
Thank you for fixed this problem. @piotr-zyskowski-rai
I close this issue.
Please re-open issue if necessary.

closed 2023-07-14 by hakuturu583