tier4/scenario_simulator_v2 #943: Unable to install scenario_simulator_v2 because rosdep fails to resolve dependencies

AndriiChumak145, 2022-12-22:
**Describe the bug**
I've tried to install the simulator as described here: [https://autowarefoundation.github.io/autoware-documentation/main/tutorials/scenario-simulation/planning-simulation/installation/)]()

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to autoware workspace
`cd autoware`
2. Import Simulator dependencies:
`vcs import src < simulator.repos`
3. Install dependent ROS packages:
```
source /opt/ros/galactic/setup.bash
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
```
Note: in the documentaion, they are using ros _humble_ instead of _galactic_

**Expected behavior**
rosdep is supposed to install the needed dependencies, but rosdep keys could not be resolved:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
tracetools_launch: Cannot locate rosdep definition for [ament_mypy]
behavior_tree_plugin: Cannot locate rosdep definition for [quaternion_operation]
autoware_auto_geometry: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
concealer: Cannot locate rosdep definition for [ament_cmake_clang_format]
random_test_runner: Cannot locate rosdep definition for [ament_cmake_clang_format]
tracetools_test: Cannot locate rosdep definition for [ament_cmake_mypy]
tracetools_trace: Cannot locate rosdep definition for [ament_mypy]
traffic_simulator_msgs: Cannot locate rosdep definition for [ament_cmake_clang_format]
simulation_interface: Cannot locate rosdep definition for [zmqpp_vendor]
ros2trace: Cannot locate rosdep definition for [ament_mypy]
tum_trajectory_planner_emulator: Cannot locate rosdep definition for [matplotlib_cpp]
traffic_simulator: Cannot locate rosdep definition for [quaternion_operation]
geometry: Cannot locate rosdep definition for [quaternion_operation]
user_defined_value_condition_example: Cannot locate rosdep definition for [ament_cmake_clang_format]
simple_junit: Cannot locate rosdep definition for [ament_cmake_clang_format]
openscenario_preprocessor: Cannot locate rosdep definition for [ament_cmake_clang_format]
kashiwanoha_map: Cannot locate rosdep definition for [ament_cmake_clang_format]
openscenario_interpreter: Cannot locate rosdep definition for [ament_cmake_clang_format]
openscenario_visualization: Cannot locate rosdep definition for [color_names]
radar_threshold_filter: Cannot locate rosdep definition for [ament_clang_format]
tracetools_read: Cannot locate rosdep definition for [ament_mypy]
openscenario_interpreter_example: Cannot locate rosdep definition for [ament_cmake_clang_format]
tum_control_visualizer: Cannot locate rosdep definition for [matplotlib_cpp]
scenario_simulator_exception: Cannot locate rosdep definition for [ament_cmake_clang_format]
cpp_mock_scenarios: Cannot locate rosdep definition for [ament_cmake_clang_format]
simple_sensor_simulator: Cannot locate rosdep definition for [quaternion_operation]
```

Because of the issues with rosdep, I'm not able to build and use the scenario_test_runner

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - ROS2 galactic
 - autoware.universe, scenario_simulator_v2

HansRobo, 2022-12-23:
@AndriiChumak145 
Thank you for reporting!
The error seems to occur due to the EOL of Galactic.

Would you mind trying to execute below commands instead of step 3?

```bash
source /opt/ros/galactic/setup.bash
rosdep update --include-eol-distros
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
```

AndriiChumak145, 2022-12-23:
Thanks, your suggestion helped to resolve the rosdep issue. After that, I ran `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release --packages-up-to scenario_test_runner`  to build the needed packages and got the following error:
```
--- stderr: geometry                                                                                                                                                                                       
make[2]: Warning: File 'gtest/CMakeFiles/gtest_main.dir/depend.make' has modification time 1233 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File 'gtest/CMakeFiles/gtest_main.dir/depend.make' has modification time 1233 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File 'CMakeFiles/geometry.dir/src/intersection/intersection.cpp.o' has modification time 1235 s in the future
/home/andrii/autoware/src/simulator/scenario_simulator/common/math/geometry/src/polygon/polygon.cpp:15:10: fatal error: quaternion_operation/quaternion_operation.h: No such file or directory
   15 | #include <quaternion_operation/quaternion_operation.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

I tried to solve the problem and installed quaternion_operation package myself. This solved the problem with geometry but after that I got:
```
--- stderr: simulation_interface                            
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Protobuf_LIBRARY
    linked by target "simulation_interface" in directory /home/andrii/autoware/src/simulator/scenario_simulator/simulation/simulation_interface

---
```
As I understood, I need the protobuf library. I tried to install it from https://www.datascienceblog.net/post/programming/essential-protobuf-guide-python/  but the installation wasn't successful:
```
protoc
protoc: error while loading shared libraries: libprotoc.so.23: cannot open shared object file: No such file or directory
```
I also found a line in CMake `include(FindProtobuf REQUIRED)`. Does the project need only FindProtobuf file?

HansRobo, 2022-12-23:
`FindProtobuf` should be provided by CMake as standard.
ref: https://github.com/Kitware/CMake/blob/master/Modules/FindProtobuf.cmake

Can you check if the packages below are installed?
They should be resolved and installed by rosdep, but they seem to be missing.

- libprotobuf-dev
- protobuf-compiler
- libprotoc-dev
- libprotobuf17

AndriiChumak145, 2022-12-23:
I've checked the packages and only the last one was installed, so I've also installed the three above. I've also run `rosdep install` explicitly on _src/simulator_ and it installed all the required rosdeps.
Now, the build fails with a different error:
```
--- stderr: simulation_interface                            
CMake Error at /usr/share/cmake-3.16/Modules/FindProtobuf.cmake:469 (file):
  file STRINGS file "/usr/local/include/google/protobuf/stubs/common.h"
  cannot be read.
Call Stack (most recent call first):
  CMakeLists.txt:22 (find_package)

CMake Error at /usr/share/cmake-3.16/Modules/FindProtobuf.cmake:475 (math):
  math cannot parse the expression: " / 1000000": syntax error, unexpected
  exp_DIVIDE (2).
Call Stack (most recent call first):
  CMakeLists.txt:22 (find_package)

CMake Error at /usr/share/cmake-3.16/Modules/FindProtobuf.cmake:476 (math):
  math cannot parse the expression: " / 1000 % 1000": syntax error,
  unexpected exp_DIVIDE (2).
Call Stack (most recent call first):
  CMakeLists.txt:22 (find_package)

CMake Error at /usr/share/cmake-3.16/Modules/FindProtobuf.cmake:477 (math):
  math cannot parse the expression: " % 1000": syntax error, unexpected
  exp_MOD (2).
Call Stack (most recent call first):
  CMakeLists.txt:22 (find_package)
```

hakuturu583, 2022-12-26:
This is the first error I have seen.
Can you give me a version of protobuf?
Also, I would appreciate it if you could try a clean build.

AndriiChumak145, 2022-12-29:
My protobuf version is 3.6.1
I've cleaned the build and install space, reinstalled dependencies, and tried to rebuild everything. Now, it fails on following:
```
--- stderr: lanelet2_matching                                                                                                                                                                                     
CMake Error at /opt/ros/galactic/share/mrt_cmake_modules/cmake/Modules/GatherDeps.cmake:24 (message):
  Gather depenencies failed: Traceback (most recent call last):

    File "/opt/ros/galactic/share/mrt_cmake_modules/cmake/../scripts/generate_cmake_dependency_file.py", line 366, in <module>
      main(sys.argv[1], sys.argv[2], sys.argv[3])
    File "/opt/ros/galactic/share/mrt_cmake_modules/cmake/../scripts/generate_cmake_dependency_file.py", line 293, in main
      catkin_packages = getCatkinPackages(workspaceRoot)
    File "/opt/ros/galactic/share/mrt_cmake_modules/cmake/../scripts/generate_cmake_dependency_file.py", line 268, in getCatkinPackages
      getPackagesInPath(packages, path)
    File "/opt/ros/galactic/share/mrt_cmake_modules/cmake/../scripts/generate_cmake_dependency_file.py", line 234, in getPackagesInPath
      package = ET.parse(os.path.join(root, manifest)).getroot()
    File "/usr/lib/python3.8/xml/etree/ElementTree.py", line 1202, in parse
      tree.parse(source, parser)
    File "/usr/lib/python3.8/xml/etree/ElementTree.py", line 584, in parse
      source = open(source, "rb")

  FileNotFoundError: [Errno 2] No such file or directory:
  '/home/andrii/autoware/install/simple_planning_simulator/share/simple_planning_simulator/package.xml'

Call Stack (most recent call first):
  CMakeLists.txt:11 (include)
```
11. line in the CMakeLists.txt: `include(GatherDeps)`

HansRobo, 2023-02-15:
> @AndriiChumak145 What rosdistro and Linux distribution are you using?

@hakuturu583 
To his credit, he already wrote it.
Please see his first post.

HansRobo, 2023-02-15:
> @HansRobo Can you check protoc version?

My version of protobuf is also `3.12.4` because I'm using Ubuntu 22.04.
And `3.6.1` is fomal version for Ubuntu 20.04.

hakuturu583, 2023-07-31:
Humm..
It seems a problem with protoc version.
But I think this tool is still work with galactic.
I will check the protoc version in github actions environment.

stevenbrills, 2023-08-22:
Has there been any updates on this front? I too am trying to compile this package on ROS2 galactic with Autoware.Universe. I've resolved the quaternion_operation dependency by compiling it from source and adding it to the package path. Now zeromq is said to be missing while compiling, specifically from the simulation_interface package. But when running rosdep for simulation_interface I get that all dependencies are installed. The manifest lists "zmqpp_vendor" as the dependency and rosdep reports that it is installed but build returns a message saying "zmqpp/zmqpp.hpp" is missing.

Can I get some insight regarding this?

hakuturu583, 2023-12-04:
@stevenbrills Galactic support is already dropped and zmqpp_vendor package is released on [humble.](https://index.ros.org/p/zmqpp_vendor/)

hakuturu583, 2024-02-14:
I think this issue is no longer active, if necessary, please re-open issue.

closed 2024-02-14 by hakuturu583