tier4/scenario_simulator_v2 #974: Simulator loads the scenario but does not run

VRichardJP, 2023-03-08:
**Describe the bug**
I am using latest autoware code and followed the instructions here to setup the simulator: https://tier4.github.io/scenario_simulator_v2-docs/user_guide/QuickStart/

Whatever the scenario I try to run, the simulation simply does not move at all and eventually timeout.

For example, when I run one of the ODD WG scenarios as such:
```
$ ros2 launch scenario_test_runner scenario_test_runner.launch.py scenario:=/home/sig/autoware/src/simulator/operational-design-domains/cargo_delivery/scenarios/UC-011-0002-Kashiwa.yaml sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle launch_rviz:=true record:=false global_frame_rate:=5.0
```

The scenario initial position seems ready:
[image]

But nothing ever happens and the simulation is stopped after the timeout.

Here are the logs from the simulation above:
[scenario_test_runner.txt](https://github.com/tier4/scenario_simulator_v2/files/10919198/scenario_test_runner.txt)

**To Reproduce**
```
$ ros2 launch scenario_test_runner scenario_test_runner.launch.py scenario:=/home/sig/autoware/src/simulator/operational-design-domains/cargo_delivery/scenarios/UC-011-0002-Kashiwa.yaml sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle launch_rviz:=true record:=false global_frame_rate:=5.0
```

**Expected behavior**
Something moves?

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 (docker)
 - Browser: N/A
 - Version: latest main branch
 - ROS2 version: humble
 - DDS: cycloneDDS

hakuturu583, 2023-03-09:
Thank you for posting issue.
I checked your log and it seems there are many logs like below.

```
Your machine is not powerful enough to run the scenario at the specified frame rate (5 Hz). We recommend that you reduce the frame rate to 1.64745 or less.
```

Can you tell me your PC spec?

VRichardJP, 2023-03-09:
Intel i7-9700 (8) @ 4.700GHz
NVIDIA Quadro P4000

VRichardJP, 2023-03-09:
I just tested again now. It worked once (the vehicles moved), but after that I was not able to run the simulation again. I have realized that the simulation quits earlier than the 5 minutes timeout, rather something like after 1 minute. The logs seems to show the openscenario node is killing autoware after just 30 seconds :
```$ ros2 launch scenario_test_runner scenario_test_runner.launch.py scenario:=/home/sig/autoware/src/simulator/operational-design-domains/cargo_delivery/scenarios/UC-011-0002-Kashiwa.yaml sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle launch_rviz:=true record:=false global_frame_rate:=5.0[INFO] [launch]: All log files can be found below /home/sig/.ros/log/2023-03-09-07-04-46-066596-vr-desktop-280078
[INFO] [launch]: Default logging verbosity is set to INFO
architecture_type       := awf/universe
autoware_launch_file    := planning_simulator.launch.xml
autoware_launch_package := autoware_launch
global_frame_rate       := 5.0
global_real_time_factor := 1.0
global_timeout          := 180
initialize_duration     := 30
launch_autoware         := True
launch_rviz             := true
output_directory        := /tmp
port                    := 8080
record                  := false
rviz_config             := 
scenario                := /home/sig/autoware/src/simulator/operational-design-domains/cargo_delivery/scenarios/UC-011-0002-Kashiwa.yaml
sensor_model            := sample_sensor_kit
sigterm_timeout         := 8
vehicle_model           := sample_vehicle
workflow                := /dev/null
[INFO] [scenario_test_runner.py-1]: process started with pid [280091]
[INFO] [simple_sensor_simulator_node-2]: process started with pid [280093]
[INFO] [openscenario_interpreter_node-3]: process started with pid [280095]
[INFO] [openscenario_preprocessor_node-4]: process started with pid [280097]
[INFO] [openscenario_visualization_node-5]: process started with pid [280099]
[INFO] [rviz2-6]: process started with pid [280101]
[scenario_test_runner.py-1] [INFO 1678345486.780469145] [simulation.scenario_test_runner]: connection established with preprocessor (print_debug() at /home/sig/autoware/install/scenario_test_runner/lib/scenario_test_runner/scenario_test_runner.py:301)
[scenario_test_runner.py-1] [INFO 1678345488.133500574] [simulation.scenario_test_runner]: Result of /simulation/openscenario_preprocessor/load: success (print_debug() at /home/sig/autoware/install/scenario_test_runner/lib/scenario_test_runner/scenario_test_runner.py:301)
...
...
...
[openscenario_interpreter_node-3] [component_container_mt-41] [INFO 1678345518.680590796] [default_ad_api.node.localization]: client exit: /localization/initialize (log() at /home/sig/autoware/install/component_interface_utils/include/component_interface_utils/rclcpp/interface.hpp:55)
[openscenario_interpreter_node-3] [component_container_mt-41] [INFO 1678345518.681211482] [default_ad_api.node.localization]: server exit: /api/localization/initialize (log() at /home/sig/autoware/install/component_interface_utils/include/component_interface_utils/rclcpp/interface.hpp:55)
[openscenario_interpreter_node-3] [initial_pose_adaptor-43] [INFO 1678345518.681446514] [default_ad_api.helpers.initial_pose_adaptor]: client exit: /api/localization/initialize (log() at /home/sig/autoware/install/component_interface_utils/include/component_interface_utils/rclcpp/interface.hpp:55)
[openscenario_interpreter_node-3] [component_container_mt-27] [INFO 1678345518.758539191] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner]: waiting for scenario_topic (operator()() at /home/sig/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/behavior_path_planner_node.cpp:800)
[openscenario_interpreter_node-3] [component_container-32] [INFO 1678345518.857062057] [control.trajectory_follower.controller_node_exe]: Waiting for trajectory. (createInputData() at /home/sig/autoware/src/universe/autoware.universe/control/trajectory_follower_node/src/controller_node.cpp:150)
[openscenario_interpreter_node-3] [component_container-32] [INFO 1678345518.857104932] [control.trajectory_follower.controller_node_exe]: Control is skipped since input data is not ready. (callbackTimerControl() at /home/sig/autoware/src/universe/autoware.universe/control/trajectory_follower_node/src/controller_node.cpp:189)
[openscenario_interpreter_node-3] [INFO 1678345519.353117597] [simulation.concealer]: Simulator waiting for Autoware state to be WaitingForRoute (remain: 0). (waitForAutowareStateToBeWaitingForRoute() at /home/sig/autoware/src/simulator/scenario_simulator_v2/external/concealer/include/concealer/transition_assertion.hpp:120)
[openscenario_interpreter_node-3] [initial_pose_adaptor-43] [INFO 1678345519.354556926] [default_ad_api.helpers.initial_pose_adaptor]: map fit1: 3747.885 73738.815 0.174 (fit() at /home/sig/autoware/src/universe/autoware.universe/map/map_height_fitter/src/map_height_fitter.cpp:169)
[openscenario_interpreter_node-3] [initial_pose_adaptor-43] [INFO 1678345519.363513588] [default_ad_api.helpers.initial_pose_adaptor]: Send request to map_loader (get_partial_point_cloud_map() at /home/sig/autoware/src/universe/autoware.universe/map/map_height_fitter/src/map_height_fitter.cpp:103)
[openscenario_interpreter_node-3] [initial_pose_adaptor-43] [INFO 1678345519.535547323] [default_ad_api.helpers.initial_pose_adaptor]: Loaded partial pcd map from map_loader (grid size: 1) (get_partial_point_cloud_map() at /home/sig/autoware/src/ [... cut ...] ulator/scenario_simulator_v2/external/concealer/src/autoware.cpp:98)
[openscenario_interpreter_node-3] [INFO 1678345520.448689025] [simulation.concealer]: Shutting down Autoware: (2/3) Terminating Autoware. (shutdownAutoware() at /home/sig/autoware/src/simulator/scenario_simulator_v2/external/concealer/src/autoware.cpp:103)
[openscenario_interpreter_node-3] [WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
```

Between the command start and the "Shutting down", there are only 33 seconds.

VRichardJP, 2023-03-09:
I think I have found the "cause" of the issue. After a first simulation, many ros related programs are not shut down properly:
```
ps aux | grep ros
sig       284570  1.3  0.1 596712 45396 pts/1    Sl   07:13   0:00 /opt/ros/humble/lib/robot_state_publisher/robot_state_publisher --ros-args -r __node:=robot_state_publisher -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7 -p coordinate_system:=4 -p utm_zone:=54 -p mgrs_zone:=54SUE -p jp_plane:=9 -p latitude:=40.81187906 -p longitude:=29.3581011 -p altitude:=47.62 --params-file /tmp/launch_params_y2xu9d2q
sig       284572  1.3  0.1 593756 41652 pts/1    Sl   07:13   0:00 /home/sig/autoware/install/component_interface_tools/lib/component_interface_tools/service_log_checker --ros-args -r __node:=service_log_checker -r __ns:=/system -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7 -p coordinate_system:=4 -p utm_zone:=54 -p mgrs_zone:=54SUE -p jp_plane:=9 -p latitude:=40.81187906 -p longitude:=29.3581011 -p altitude:=47.62
sig       284574  3.2  0.1 595336 43792 pts/1    Sl   07:13   0:01 /opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=container -r __ns:=/system/component_state_monitor -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7 -p coordinate_system:=4 -p utm_zone:=54 -p mgrs_zone:=54SUE -p jp_plane:=9 -p latitude:=40.81187906 -p longitude:=29.3581011 -p altitude:=47.62
sig       284576  1.4  0.1 594052 42260 pts/1    Sl   07:13   0:00 /home/sig/autoware/install/topic_state_monitor/lib/topic_state_monitor/topic_state_monitor_node --ros-args -r __node:=topic_state_monitor_vector_map -r __ns:=/system -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7 -p coordinate_system:=4 -p utm_zone:=54 -p mgrs_zone:=54SUE -p jp_plane:=9 -p latitude:=40.81187906 -p longitude:=29.3581011 -p altitude:=47.62 --params-file /tmp/launch_params_rnkei6u3 --params-file /tmp/launch_params_viv2os5t --params-file /tmp/launch_params_6i3l_wof --params-file /tmp/launch_params_7wlsfa6f --params-file /tmp/launch_params_pbet40mm --params-file /tmp/launch_params_pzt9iexr --params-file /tmp/launch_params_hdb8ujao --params-file /tmp/launch_params_6cbo3vqq --params-file /tmp/launch_params_d5tyb7so
sig       284578  1.4  0.1 594044 41856 pts/1    Sl   07:13   0:00 /home/sig/autoware/install/topic_state_monitor/lib/topic_state_monitor/topic_state_monitor_node --ros-args -r __node:=topic_state_monitor_initialpose3d -r __ns:=/system -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7 -p coordinate_system:=4 -p utm_zone:=54 -p mgrs_zone:=54SUE -p jp_plane:=9 -p latitude:=40.81187906 -p longitude:=29.3581011 -p altitude:=47.62 --params-file /tmp/launch_params_dva4k5v1 --params-file /tmp/launch_params_ltaupgab --params-file /tmp/launch_params_yfz6vyci --params-file /tmp/launch_params_bc5k8pjc --params-file /tmp/launch_params_lpzan8zq --params-file /tmp/launch_params_1yt948zd --params-file /tmp/launch_params_76c556o_ --params-file /tmp/launch_params_2evac2rw --params-file /tmp/launch_params_088jdx6k
sig       284580  1.5  0.1 595028 43240 pts/1    Sl   07:13   0:00 /home/sig/autoware/install/topic_state_monitor/lib/topic_state_monitor/topic_state_monitor_node --ros-args -r __node:=topic_state_monitor_object_recognition_objects -r __ns:=/system -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7 -p coordinate_system:=4 -p utm_zone:=54 -p mgrs_zone:=54SUE -p jp_plane:=9 -p latitude:=40.81187906 -p longitude:=29.3581011 -p altitude:=47.62 --params-file /tmp/launch_params_8kpj1tuh --params-file /tmp/launch_params_yil55tcf --params-file /tmp/launch_params_v9h99sdm --params-file /tmp/launch_params_oog3w9hf --params-file /tmp/launch_params_aanr8ou6 --params-file /tmp/launch_params_0nodjx5e --params-file /tmp/launch_params_4abmb4py --params-file /tmp/launch_params_uzv0zgmb --params-file /tmp/launch_params_lnz5r9y1
sig       284582  1.4  0.1 594332 42148 pts/1    Sl   07:13   0:00 /home/sig/autoware/install/topic_state_monitor/lib/topic_state_monitor/topic_state_monitor_node --ros-args -r __node:=topic_state_monitor_mission_planning_route -r __ns:=/system -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:= [... cut ...]
If I kill everything then run the scenario_test_runner , it works:
```
$ ps aux | grep ros | head -n -1 | awk '{ print $2 }' | xargs kill -9
$ ros2 launch scenario_test_runner scenario_test_runner.launch.py scenario:=/home/sig/autoware/src/simulator/operational-design-domains/cargo_delivery/scenarios/UC-011-0002-Kashiwa.yaml sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle launch_rviz:=false record:=false global_frame_rate:=5.0
```

[image]

After the scenario_test_runner has finished the simulation and quit, the problem occurs again.

hakuturu583, 2023-03-16:
I have had the same symptoms.
Indeed, the self-diagnostic node survived after the end of the simulation, and this caused a situation where the simulation could not be performed normally.
I have confirmed that this also occurs in other people's environments.

hakuturu583, 2023-07-18:
@VRichardJP Do you still have this problem?

VRichardJP, 2023-07-18:
I have not been using the simulator since then, so I don't know. If no one is experiencing the problem anymore we can close it. It is most likely environment related.

hakuturu583, 2023-07-20:
Thanks!
I close this pull-request.
If someone face same problem, I will re-open this pull-request.

closed 2023-07-20 by hakuturu583

jxa124, 2023-09-28:
> Thanks! I close this pull-request. If someone face same problem, I will re-open this pull-request.

I met this bug too