tier4/scenario_simulator_v2 #987: Scenario Editor fails to load the sample maps in the autoware planning tutorial.

hcn1519, 2023-04-05:
**Describe the bug**
The scenario editor cannot load the sample maps from the tutorial in the autoware documentation of the [tutorial](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/).

**To Reproduce**
Steps to reproduce the behavior:

I was trying to load the map following https://tier4.github.io/scenario_simulator_v2-docs/user_guide/scenario_editor/CreateSimpleScenario/

1. Go to [Tutorials](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/) in Autoware documentation
2. Download sample map planning
3. Click load map from the scenario editor 
4. Choose `lanelet2_map.osm` file

**Expected behavior**
Sample map should be loaded 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: ubuntu
 - Browser Chrome
 - Version 111.0.5563.64 (Official Build) (64-bit)
 - URL: https://scenario.ci.tier4.jp/scenario_editor/

hcn1519, 2023-04-05:
I have another finding. I can load the map in this repository([kashiwanoha_map](https://github.com/tier4/scenario_simulator_v2/tree/master/map/kashiwanoha_map/map))

hakuturu583, 2023-07-19:
Sorry for my late reply.
I will ask to developer of ScenarioEditor.

hakuturu583, 2023-07-20:
I can import [the map you specified](https://drive.google.com/file/d/1499_nsbUbIeturZaDj7jhUownh5fvXHd/view) in this issue on the scenario editor.
Do you still have problem?

hakuturu583, 2023-07-31:
I will close this issue.
If you still have problems, please re-open issue.