tier4/scenario_simulator_v2 #991: Cannot execute random_test_runner in Quick Start

hcn1519, 2023-04-14:
**Describe the bug**

The `random_test_runner` example in https://tier4.github.io/scenario_simulator_v2-docs/user_guide/QuickStart/ does not work. It stops running after outputting the log below.

```
[INFO] [launch]: All log files can be found below /home/sora/.ros/log/2023-04-14-14-08-02-939099-45608c4816fb-105885
[INFO] [launch]: Default logging verbosity is set to INFO
Value of simulator_type: ['simple_sensor_simulator', 'unity']
Value of architecture_type: ['awf/auto', 'awf/universe', 'tier4/proposal']
Test param file ''
Autoware architecture 'awf/universe'
Vehicle info parameters file supplied: '/home/sora/Desktop/changnam/autoware/install/sample_vehicle_description/share/sample_vehicle_description/config/vehicle_info.param.yaml'. Parameters passed there override passed via arguments
Simulator model parameters file supplied: '/home/sora/Desktop/changnam/autoware/install/sample_vehicle_description/share/sample_vehicle_description/config/simulator_model.param.yaml'. Parameters passed there override passed via arguments
Chosen simulation type simple_sensor_simulator
[INFO] [random_test_runner-1]: process started with pid [105886]
[INFO] [openscenario_visualization_node-2]: process started with pid [105888]
[INFO] [simple_sensor_simulator_node-3]: process started with pid [105890]
[random_test_runner-1] [INFO] [1681506483.072895713] [simulation.random_test_runner]: test control parameters: input dir:  output dir: /tmp random test type: 0 test count 5
[random_test_runner-1] [INFO] [1681506483.073738632] [simulation.random_test_runner]: Map path found: /home/sora/Desktop/changnam/autoware/install/kashiwanoha_map/share/kashiwanoha_map/map
[random_test_runner-1] [WARN] [1681506483.086263345] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[random_test_runner-1] [INFO] [1681506483.105058122] [simulation.random_test_runner]: Test suite parameters ego_goal_lanelet_id: -1 ego_goal_s: 0 ego_goal_partial_randomization: false ego_goal_partial_randomization_distance: 20 npcs_count: 10 npc_min_speed: 0.5 npc_max_speed: 3 npc_min_spawn_distance_from_ego: 10 npc_max_spawn_distance_from_ego: 100 name: random_test map_name: kashiwanoha_map
[random_test_runner-1] [INFO] [1681506483.105072545] [simulation.random_test_runner]: Test case 0 parameters:
[random_test_runner-1] [INFO] [1681506483.105074752] [simulation.random_test_runner]: Test case 1 parameters:
[random_test_runner-1] [INFO] [1681506483.105076267] [simulation.random_test_runner]: Test case 2 parameters:
[random_test_runner-1] [INFO] [1681506483.105077710] [simulation.random_test_runner]: Test case 3 parameters:
[random_test_runner-1] [INFO] [1681506483.105079160] [simulation.random_test_runner]: Test case 4 parameters:
[random_test_runner-1] [INFO] [1681506483.105134093] [simulation.random_test_runner]: Generating test 1/5
[random_test_runner-1] [INFO] [1681506483.105149797] [simulation.random_test_runner]: Goal randomization: full
[random_test_runner-1] [INFO] [1681506483.105582497] [simulation.random_test_runner]: Generating test 2/5
[random_test_runner-1] [INFO] [1681506483.105590329] [simulation.random_test_runner]: Goal randomization: full
[random_test_runner-1] [INFO] [1681506483.105835507] [simulation.random_test_runner]: Generating test 3/5
[random_test_runner-1] [INFO] [1681506483.105841546] [simulation.random_test_runner]: Goal randomization: full
[random_test_runner-1] [INFO] [1681506483.106099648] [simulation.random_test_runner]: Generating test 4/5
[random_test_runner-1] [INFO] [1681506483.106105283] [simulation.random_test_runner]: Goal randomization: full
[random_test_runner-1] [INFO] [1681506483.106128312] [simulation.random_test_runner]: Goal randomization: full
[random_test_runner-1] [INFO] [1681506483.106150004] [simulation.random_test_runner]: Goal randomization: full
[random_test_runner-1] [INFO] [1681506483.106412453] [simulation.random_test_runner]: Generating test 5/5
[random_test_runner-1] [INFO] [1681506483.106417573] [simulation.random_test_runner]: Goal randomization: full
[random_test_runner-1] [INFO] [1681506483.106562791] [simulation.random_test_runner]: Goal randomization: full
[random_test_runner-1] [INFO] [1681506483.106709741] [simulation.random_test_runner]: Saving yaml: /tmp/result.yaml
[random_test_runner-1] [INFO] [1681506483.106888924] [simulation.random_test_runner]: Running test 1/5
[random_test_runner-1] [INFO] [1681506483.106906710] [simulation.random_test_runner]: Test description: ego_start_position: lanelet_id: 34624, s: 3.885149019675154, offset: 0, rpy: (x, y, z) : (0, 0, 0) ego_goal_position: lanelet_id: 34639, s: 13.086546456049982, offset: 0, rpy: (x, y, z) : (0, 0, 0) ego_goal_pose: position: (x, y, z), (3769.3768181255277, 73738.9270078614, 0.07161178278169551), orientation (x, y, z, w), (0, 0, -0.5569801741840686, 0.830525788621813)
[random_test_runner-1] npc_descriptions:[0]: name: npc0, start_position: lanelet_id: 34630, s: 23.569709429068908, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 0.9980075197576856
[random_test_runner-1] [1]: name: npc1, start_position: lanelet_id: 34630, s: 16.38003006311335, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 0.5215638559921278
[random_test_runner-1] [2]: name: npc2, start_position: lanelet_id: 34750, s: 6.934661459725882, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 0.8901275490658923
[random_test_runner-1] [3]: name: npc3, start_position: lanelet_id: 34513, s: 3.1029740903992753, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 2.157664915425591
[random_test_runner-1] [4]: name: npc4, start_position: lanelet_id: 34756, s: 7.838646975061262, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 1.1185679940061197
[random_test_runner-1] [5]: name: npc5, start_position: lanelet_id: 34513, s: 30.811385665318046, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 2.8581797627738803
[random_test_runner-1] [6]: name: npc6, start_position: lanelet_id: 34741, s: 19.20998739170568, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 0.8811421721631539
[random_test_runner-1] [7]: name: npc7, start_position: lanelet_id: 34696, s: 15.243454649871111, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 1.6900407757343527
[random_test_runner-1] [8]: name: npc8, start_position: lanelet_id: 34741, s: 34.336355561987574, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 2.087731655672398
[random_test_runner-1] [9]: name: npc9, start_position: lanelet_id: 34498, s: 4.331667491631521, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 1.5122502555986066
[random_test_runner-1] 
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build `Autoware` with `scenario_simulator_v2` in docker environment.
2. Execute below command 

```shell
ros2 launch random_test_runner random_test.launch.py \
architecture_type:=awf/universe \
sensor_model:=sample_sensor_kit \
vehicle_model:=sample_vehicle
```

**Expected behavior**
- Random test runner should run randomly generated test scenarios in simulator.
- I can run scenarios using `scenario_test_runner`, and `cpp scenario demo`

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: ubuntu 20.04
 - Docker image - ghcr.io/autowarefoundation/autoware-universe:latest-cuda
 - ROS2 version - humble

piotr-zyskowski-rai, 2023-04-25:
@hcn1519 Thank you for the report! I was able to reproduce the issue, and I will be looking into it.

Zoe-Wan, 2023-05-05:
Yes, I have met this problem either. I'm using WSL2 Ubuntu 20.04, with the same version of docker image and ROS2, I can't start the rviz.

hakuturu583, 2024-02-14:
I think this issue is no longer active, please re-open issues if necessary.

closed 2024-02-14 by hakuturu583