tier4/scenario_simulator_v2 #995: please refer offisial documentation of ScenarioEditor

HansRobo, 2023-04-26:
**Is your feature request related to a problem? Please describe.**

scenario_simulator_v2 documentation includes (outdated?) ScenarioEditor documentation

**Describe the solution you'd like**

refer [Web.Auto documentation](https://docs.web.auto/user-manuals/scenario-editor/introduction)

hakuturu583, 2023-07-14:
I think it is a good idea, but this document is japanese.
I will search english version.

hakuturu583, 2023-07-14:
I found [it.](https://docs.web.auto/en/user-manuals/scenario-editor/introduction)

hakuturu583, 2023-07-18:
@HansRobo I did not check well, but it seems some of the documentation was deleted.

HansRobo, 2023-07-19:
@HansRobo 
I cannot judge whether "some of the documentation was deleted" itself is bad or not.
But, I was advised to replace it by TIER IV CI/CD team, so I think it's better to replace it.

HansRobo, 2023-07-19:
@hakuturu583 
I cannot judge whether "some of the documentation was deleted" itself is bad or not.
But, I was advised to replace it by TIER IV CI/CD team, so I think it's better to replace it.

hakuturu583, 2023-07-19:
@HansRobo I agree.
I will ask to CI/CD team.

HansRobo, 2023-07-19:
close with #1036

closed 2023-07-19 by HansRobo