tier4/scenario_simulator_v2 #1015: scenario_test_runner seems not killing rviz

hayato-m126, 2023-06-07:
**Describe the bug**

When I launch a scenario with scenario_test_runner, multiple rviz processes remain.
Running the following command will leave 3 rviz processes.

https://tier4.github.io/scenario_simulator_v2-docs/user_guide/QuickStart/#scenario_test_runner

```shell
ros2 launch scenario_test_runner scenario_test_runner.launch.py architecture_type:=awf/universe record:=false scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml' sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle
```

**To Reproduce**
Steps to reproduce the behavior:
1. build latest autoware foundation's autoware
2. ros2 launch scenario_test_runner scenario_test_runner.launch.py architecture_type:=awf/universe record:=false scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml' sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle
3. run process monitor

**Expected behavior**
Rviz process is killed properly every time.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - Browser Chrome
 - Version [e.g. 22]
 - ROS 2 Humble
 - DDS Cyclone DDS

villaoningfootball, 2023-07-13:
I also encountered the same problem. Please how to solve? thank you

hayato-m126, 2023-07-30:
A tentative solution is to increase the value of `timeout.tv_sec`

https://github.com/tier4/scenario_simulator_v2/blob/master/external/concealer/src/field_operator_application.cpp#L103

hakuturu583, 2023-07-31:
Do you have any idea of this behaviour?
I git blamed the relevant business and it seems that it was last edited by @yamacir-kit.

https://github.com/tier4/scenario_simulator_v2/blame/master/external/concealer/src/field_operator_application.cpp#L103

yamacir-kit, 2023-07-31:
I know how to fix it, but I don't have time to fix it.

villaoningfootball, 2023-08-07:
> I know how to fix it, but I don't have time to fix it.

we need you

closed 2024-02-08 by yamacir-kit