tier4/scenario_simulator_v2 #1017: The number of tests in the result.junit.xml seems incorrect.

hayato-m126, 2023-06-09:
**Describe the bug**
I launched the scenario_test_runner according to the [QuickDemo](https://tier4.github.io/scenario_simulator_v2-docs/user_guide/QuickStart/#scenario_test_runner)

The resulting resutls.junit.xml appears as follows:

```xml
<?xml version="1.0"?>
<testsuites name="/tmp/scenario_test_runner" failures="0" errors="0" tests="0">
  <testsuite name="sample" failures="0" errors="0" tests="0">
    <testcase name="sample_2" />
    <testcase name="sample_1" />
    <testcase name="sample_0" />
  </testsuite>
</testsuites>
```

While I'm not familiar with the exact format of JUnit, based on my research, it seems that 'tests' represents the number of executed tests. Therefore, wouldn't it be strange if it doesn't appear as follows?

```xml
<?xml version="1.0"?>
<testsuites name="/tmp/scenario_test_runner" failures="0" errors="0" tests="1">
  <testsuite name="sample" failures="0" errors="0" tests="3">
    <testcase name="sample_2" />
    <testcase name="sample_1" />
    <testcase name="sample_0" />
  </testsuite>
</testsuites>
```

**To Reproduce**
Steps to reproduce the behavior:
1. build awf's autoware
2. ros2 launch scenario_test_runner scenario_test_runner.launch.py architecture_type:=awf/universe record:=false scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml' sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle
3. open `/tmp/scenario_test_runner/result.junit.xml`

**Expected behavior**
I think below

```xml
<?xml version="1.0"?>
<testsuites name="/tmp/scenario_test_runner" failures="0" errors="0" tests="1">
  <testsuite name="sample" failures="0" errors="0" tests="3">
    <testcase name="sample_2" />
    <testcase name="sample_1" />
    <testcase name="sample_0" />
  </testsuite>
</testsuites>
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - ROS 2 Humble
 - DDS Cyclone DDS

hayato-m126, 2023-06-09:
I added sample format URL

yamacir-kit, 2023-06-09:
We have confirmed that the problem can be reproduced.
This is probably an unintentional bug, since the process of outputting result.junit.xml includes a process that makes the number of `tests` be the sum of the number of `failures`, `erros` and `pass`.

hayato-m126, 2023-06-09:
Thank you for confirming.
It is not a critical issue for the evaluation results, so it is not urgent, but it would be great if you could fix this bug

PR #1037 (2023-07-20): Fix for JUnit wrong count of testcases
## Types of PR

- [ ] New Features
- [ ] Upgrade of existing features
- [x] Bugfix

## Link to the issue

- close #1017 

## Description

JUnit output library uses length of `pass` array as a passed test count. However, interpreter does not push passed case to `pass` array.
It caused writing inconsistent result to JUnit format XML.

## How to review this PR.

As no algorithm changes, regression test seems to be optional.

## Others
closed 2023-07-20 by HansRobo