tier4/scenario_simulator_v2 #1034: Split API document and user document.

hakuturu583, 2023-07-19:
**Is your feature request related to a problem? Please describe.**
Currently, the API documentation and user documentation by Doxygen are located at the same URL, which creates significant visibility issues.

**Describe the solution you'd like**
Split [API document](https://tier4.github.io/scenario_simulator_v2-docs/package/About/) and [user document](https://tier4.github.io/scenario_simulator_v2-docs/).

**Describe alternatives you've considered**
No. If you have an idea, please write on this issue.

hakuturu583, 2023-07-20:
This issue should be started after #1019 merged.

kyoichi-sugahara, 2023-07-31:
@hakuturu583 cc: @yamacir-kit @HansRobo 

I am making a rough proposal without fully understanding the current docs directory structure. This is still in the early stages, and I would like to invite active suggestions and comments. How about making the following changes to the structure of the docs folder?

- Create folders named 'about', 'installation', 'contributing', and 'how_to_guides', as adopted in the [autoware-documentation](https://autowarefoundation.github.io/autoware-documentation/main/). 'about' would contain general information for those wanting to understand the repository at a glance, 'installation' would explain how to build, 'contributing' would provide information necessary for contributing, and 'how_to_guides' would cover topics like how to create scenarios.

- Now `developer_guide` and `user_guide` are separate, and I understand the necessity to split them due to different information required from each perspective. But the information for user may contains information necessary for developer. To avoid duplication and make it easier for document maintainers to choose which file to edit, I propose we unify them under the theme of 'package explanation'.
As for where to place these unified files, I can think of two options (if you have other suggestions, please comment).

Option 1:

Create a 'package' folder and create descriptions for each package within it. I have provided a diff in this [PR](https://github.com/tier4/scenario_simulator_v2/pull/1057) to help visualize this approach.

Option 2:

As adopted in autoware, we can create a README.md within each package and add the description for that package there. This is my understanding of the current approach in this [PR](https://github.com/tier4/scenario_simulator_v2/pull/1019).

There are detailed discussions to be had, such as how to arrange existing files, or how to lay out the path for those who want to understand how to create scenarios to smoothly access the necessary information. However, before getting into the specifics, I would like to hear your opinions on the general policy of where to place the descriptions for each package.

hakuturu583, 2023-10-31:
This feature was merged.
Thank you!

closed 2023-10-31 by hakuturu583