tier4/scenario_simulator_v2 #1062: duplicated source code of "external/lanelet2_matching"

felixf4xu, 2023-08-02:
**Describe the bug**

this project depends on lanelet2:
```xml
  <depend>lanelet2_core</depend>
  <depend>lanelet2_extension</depend>
  <depend>lanelet2_io</depend>
  <depend>lanelet2_matching</depend>
  <depend>lanelet2_projection</depend>
  <depend>lanelet2_routing</depend>
```

So usually lanelet2 is built and installed from https://github.com/fzi-forschungszentrum-informatik/Lanelet2

But in the repo, the source code of "lanelet2_matching" is included again, which leads to duplication of project names.

**To Reproduce**
Steps to reproduce the behavior:
1. set up autoware.auto workspace by `vcs` and `rosdep`
2. vcs import Lanelet2 into autoware.auto workspace
3. vcs import autoware.auto's simulator folder, then this repo is cloned into autoware.auto workspace.
4. colcon build the workspace

colcon has error because `lanelet2_matching` is included twice.

**Expected behavior**
colcon build ok

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version: 22.04
 - ROS 2 version: rolling
 - DDS: default

hakuturu583, 2023-08-03:
In my environment, and Autoware, we usually use Lanelet2 from rosdep.
But lanelet2_matching was not installed via rosdep.

Today, I found that the lanelet2_matching package can be installed via rosdep, so I will modify dependency and code.
Thanks!

hakuturu583, 2023-08-04:
I found that the `lanelet2` package was not released in ROS 2 `galactic`.
So, after the `galactic` support is dropped, #1063 will be merged in master.

felixf4xu, 2023-08-04:
Do you mean when galactic support is dropped by autoware.auto? I thought it's already dropped, am I right?
https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/

hakuturu583, 2023-08-07:
Some of our project still running on galactic.
So I could not drop galactic support now.

hakuturu583, 2023-09-20:
This problem was solved by #1063 
Thank you for notify!

closed 2023-09-20 by hakuturu583