tier4/scenario_simulator_v2 #1094: Scenario simulator can't work on arm64 architecture

oguzkaganozt, 2023-09-26:
**Describe the bug**
Embree package is not available for arm64 architecture, the scenario simulator can't work on arm64 architectures

**To Reproduce**
```
vcs import src < simulator.repos
./setup-dev-env.sh
source /opt/ros/humble/setup.bash
rosdep install -iry --from-paths src --rosdistro $ROS_DISTRO
```

**Expected behavior**
Successful Rosdep installation

 - Platform: Ubuntu22.04 - ROS2 Humble - arm64

hakuturu583, 2023-10-02:
Thank you for reporting.
ARM support is discussed at the maintainers' meeting.

hakuturu583, 2023-10-26:
Thanks, currently we are trying to send patch to embree_vendor package and trying to fix this problem.

hakuturu583, 2024-04-11:
Start adding build test on ARM CPUs using qemu.
However, integration test does not provide on ARM CPUs.

hakuturu583, 2024-04-11:
I close this pull request, but if you need integration test on ARM CPU or other features.
Please re-open this pull request.

closed 2024-04-11 by hakuturu583