tier4/scenario_simulator_v2 #1114: Unnecessary equal operators

TauTheLepton, 2023-10-24:
**Describe the bug**
The functions below generate compilation errors.
https://github.com/tier4/scenario_simulator_v2/blob/173a8c1d5e16b6e594de58d78cdb81a1d518ad82/common/math/geometry/include/geometry/linear_algebra.hpp#L53-L54
Both equal operators are ambiguous. ROS2 message types have automatically generated equal operator member function. Any usage of the aforementioned functions results in a compilation error. 
It is unnecessary to keep functions that cannot be used, so I suggest removing these functions.

_Context:_
I am adding unit tests for the `geometry` package and wanted to add tests for these functions as well, but as it turns out they cannot be used. 

**To Reproduce**
Steps to reproduce the behavior:
1. Edit any source code file or add a new one
2. Include the `linear_algebra.hpp` header file
3. Write any code that uses the equal operator
4. Try to compile
5. See error

**Expected behavior**
Using `Point` and `Vector3` equal operator should result in a correct evaluation and not in a compilation error.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]: Ubuntu
 - Browser [e.g. chrome, safari]: Firefox
 - Version [e.g. 22]: 22
 - ROS 2 version: Humble
 - DDS: CycloneDDS

hakuturu583, 2023-10-25:
> Both equal operators are ambiguous. ROS2 message types have automatically generated equal operator member function. Any usage of the aforementioned functions results in a compilation error.
It is unnecessary to keep functions that cannot be used, so I suggest removing these functions.

Thank you!
Is the ROS2 message types have automatically generated equal operator consider epsilon?

I build [common_interfaces](https://github.com/ros2/common_interfaces) package and checking automatically generated equal operator for Float64 type value, it seems it is not consider machine epsilon.

```C++
  bool operator==(const Float64_ & other) const
  {
    if (this->data != other.data) {
      return false;
    }
    return true;
  }
```

hakuturu583, 2023-10-25:
If the operator does not consider machine epsilon, I think the automatically generated operator should not be used and compilation error does not raised in my environment and CI/CD environment.

TauTheLepton, 2023-10-25:
The default operator does not consider epsilon as you said, but it can be confusing to have to use the operator explicitly.
The error is raised only when the operator is used in such a way:
```cpp
vector0 == vector1;
```
This is the most common way to use the operator I think.

I have created an example test case showcasing the problem [here](https://github.com/TauTheLepton/scenario_simulator_v2/blob/1b47e80d531fcdabc5176d6a7f77aa5a5a2285db/common/math/geometry/test/test_linear_algebra.cpp#L76-L89).

When compiling the code from [this branch](https://github.com/TauTheLepton/scenario_simulator_v2/tree/example/equal-operator-compilation-error) the error can be observed. 

I found it confusing to have to use one operator explicitly as I have not found any note about this. However if you consider this to be a correct behavior then please close this issue.

hakuturu583, 2023-10-26:
I see, I encounter error.
It seems, this operator does not used in scenario_simulator_v2 so please remove the unnecessary equal operator.

hakuturu583, 2023-12-21:
#1139 was merged

closed 2023-12-21 by hakuturu583