tier4/scenario_simulator_v2 #1115: isIntersect2D nad getIntersection2D vector bug

TauTheLepton, 2023-10-24:
**Describe the bug**
The functions `isIntersect2D` and `getIntersection2D` have implementation bug.
https://github.com/tier4/scenario_simulator_v2/blob/173a8c1d5e16b6e594de58d78cdb81a1d518ad82/common/math/geometry/src/intersection/intersection.cpp#L44-L45
https://github.com/tier4/scenario_simulator_v2/blob/173a8c1d5e16b6e594de58d78cdb81a1d518ad82/common/math/geometry/src/intersection/intersection.cpp#L77-L78
All for loops above look past the last element of the `lines` vector which causes the function to work incorrectly.
The comparison operator in the loops condition should be changed from `<=` to `<` in order to check all of the elements and not more.

_Context:_
I am adding unit tests for the `geometry` package and wanted to add tests for these functions, but the functions mentioned above produce incorrect results and I believe the mentioned for loop issues is the cause for the incorrect behavior. 

**To Reproduce**
Steps to reproduce the behavior:
1. Write any code using the functions mentioned above (the ones accepting `std::vector` as argument)
2. See the incorrect result

**Expected behavior**
Functions do produce the correct result (e.g. line from (0, 0) to (1, 1) does not intersect line from point (1, 0) to (2, 1) ).

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]: Ubuntu
 - Browser [e.g. chrome, safari]: Firefox
 - Version [e.g. 22]: 22
 - ROS 2 version: Humble
 - DDS: CycloneDDS

hakuturu583, 2023-10-25:
Thanks, it was my mistake.

hakuturu583, 2023-12-21:
#1139 was merged

closed 2023-12-21 by hakuturu583