tier4/scenario_simulator_v2 #1116: LineSegment initialization bug

TauTheLepton, 2023-10-24:
**Describe the bug**
The lambda below initializes the `end_point` with `nan` values in case when passed vector has a zero length.
https://github.com/tier4/scenario_simulator_v2/blob/173a8c1d5e16b6e594de58d78cdb81a1d518ad82/common/math/geometry/src/polygon/line_segment.cpp#L39-L46
This case should be considered e.g. with an early return that returns `start_point` in case the `vec_length` is 0. 

_Context:_
I am adding unit tests for the `geometry` package and wanted to add tests for `LineSegment` initialization, but the 0 length vector case produces incorrect results.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit any existing source code file or add a new one
2. Try to initialize `LineSegment` with a vector that has a zero length
3. Print the `end_point` to the console
4. See the incorrect values

**Expected behavior**
When initializing `LineSegment` with a vector that has a length = 0, the `end_point` should have the same values as `start_point` (because it essentially becomes a point).

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]: Ubuntu
 - Browser [e.g. chrome, safari]: Firefox
 - Version [e.g. 22]: 22
 - ROS 2 version: Humble
 - DDS: CycloneDDS

hakuturu583, 2023-10-26:
Can you use THROW_SIMULATION_ERROR if the length of the lanelet is zero?

hakuturu583, 2023-12-21:
#1139 was merged

closed 2023-12-21 by hakuturu583