tier4/scenario_simulator_v2 #1117: getMinValue and getMaxValue crash the program

TauTheLepton, 2023-10-24:
**Describe the bug**
The function `getMaxValue` and `getMinValue` have a bug. When called with an empty vector they crash the program.
This is caused by the lines below.
https://github.com/tier4/scenario_simulator_v2/blob/173a8c1d5e16b6e594de58d78cdb81a1d518ad82/common/math/geometry/src/polygon/polygon.cpp#L55
https://github.com/tier4/scenario_simulator_v2/blob/173a8c1d5e16b6e594de58d78cdb81a1d518ad82/common/math/geometry/src/polygon/polygon.cpp#L61
`std::min_element` and `std::max_element`, when called with two identical iterators (which happens when we call `begin()` and `end()` on an empty vector), return this iterator. De-referencing an `end()` iterator causes the program to crash.
This case should be considered e.g. with an early return that returns `nan` when the vector is empty, or throwing an exception.

_Context:_
I am adding unit tests for the `geometry` package and wanted to add tests for the aforementioned functions, but calling them with an empty vector causes the whole program to crash.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit any existing source code file or add a new one
2. Call `getMinValue` or `getMaxValue` with an empty vector as argument
3. Compile and execute
4. See error

**Expected behavior**
Some controlled handling of the empty vector situation like return of `nan` or throwing an exception, but not program crash.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]: Ubuntu
 - Browser [e.g. chrome, safari]: Firefox
 - Version [e.g. 22]: 22
 - ROS 2 version: Humble
 - DDS: CycloneDDS

hakuturu583, 2023-10-26:
Thanks, please use THROW_SIMULATION_ERROR macro when the length of the points are zero.
And, if the length of the points are one, return the first points.

hakuturu583, 2023-12-21:
#1139 was merged

closed 2023-12-21 by hakuturu583