tier4/scenario_simulator_v2 #1118: HermiteCurve getTrajectory wrong number of points

TauTheLepton, 2023-10-24:
**Describe the bug**
The `HermiteCurve` member function `getTrajectory` taking number of points as an argument generates a trajectory with one point more than requested. The cause is that the condition in the loop below is `<=` and should be `<`.
https://github.com/tier4/scenario_simulator_v2/blob/173a8c1d5e16b6e594de58d78cdb81a1d518ad82/common/math/geometry/src/spline/hermite_curve.cpp#L226
Changing the loop condition will require the logic to be changed, including changing the `t` calculation, so going from
```cpp
double t = static_cast<double>(i) / static_cast<double>(num_points);
```
to
```cpp
double t = static_cast<double>(i) / (static_cast<double>(num_points) - 1.0);
```

_Context:_
I am adding unit tests for the `geometry` package and wanted to add tests for the `getTrajectory` function, but it generates the wrong result.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit any existing source code file or add a new one
2. Call `HermiteCurve` member function `getTrajectory` with some number of points.
3. Print to the console how many points are actually generated
4. See the difference

**Expected behavior**
The trajectory should have as many points as were requested.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]: Ubuntu
 - Browser [e.g. chrome, safari]: Firefox
 - Version [e.g. 22]: 22
 - ROS 2 version: Humble
 - DDS: CycloneDDS

hakuturu583, 2023-10-26:
Thanks, it is a bug.
Please fix it.

hakuturu583, 2023-12-21:
#1139 was merged

closed 2023-12-21 by hakuturu583