tier4/scenario_simulator_v2 #1120: CatmullRomSpline incorrect collision point calculation

TauTheLepton, 2023-10-24:
**Describe the bug**
The `CatmullRomSpline` member function `getCollisionPointIn2D` calculates wrong values.
This member function uses its component curves member function `getCollisionPointIn2D` here:
https://github.com/tier4/scenario_simulator_v2/blob/173a8c1d5e16b6e594de58d78cdb81a1d518ad82/common/math/geometry/src/spline/catmull_rom_spline.cpp#L383
and here:
https://github.com/tier4/scenario_simulator_v2/blob/173a8c1d5e16b6e594de58d78cdb81a1d518ad82/common/math/geometry/src/spline/catmull_rom_spline.cpp#L391
and simply adds the obtained value to the sum of the preceding curves. This produces an incorrect result, because `HermiteCurve` operates on normalized lengths, meaning the s value is from range <0; 1> like in the function below.
https://github.com/tier4/scenario_simulator_v2/blob/173a8c1d5e16b6e594de58d78cdb81a1d518ad82/common/math/geometry/src/spline/hermite_curve.cpp#L122-L181
And `CatmullRomSpline` operates on absolute lengths, so not normalized ones. This means that the normalized value is added to not normalized ones - which makes the incorrect result.
Many member functions in the `HermiteCurve` class offer a `denormalize_s` argument which - when set to true - converts normalized s to the absolute value of s, independent from the curvature length.
In my opinion, the best solution would be to call the `HermiteCurve` member function `getCollisionPointIn2D` with such argument `denormalize_s` set to true. However this member function does not offer such a functionality, so I suggest adding it the same way it is implemented here:
https://github.com/tier4/scenario_simulator_v2/blob/173a8c1d5e16b6e594de58d78cdb81a1d518ad82/common/math/geometry/src/spline/hermite_curve.cpp#L194-L197

_Context:_
I am adding unit tests for the geometry package and wanted to add tests for the `getCollisionPointIn2D` function, but it generates the wrong results.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit any existing source code file or create a new one
2. Create a `CatmullRomSpline` from points spaced apart for more than 2 units (like (0, 0), (2, 0) and (4, 0) )
3. Call the `getCollisionPointIn2D` member function with two points creating a parallel line to the spline (e.g. (0.1, 1) and (0.1, -1)
4. Print the result to the console
5. Compile and execute
6. See the incorrect result

**Expected behavior**
The `CatmullRomSpline` member function `getCollisionPointIn2D` should return the correct result (in the example above 0.1, not 0.05).

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]: Ubuntu
 - Browser [e.g. chrome, safari]: Firefox
 - Version [e.g. 22]: 22
 - ROS 2 version: Humble
 - DDS: CycloneDDS

hakuturu583, 2023-10-27:
Thanks, it is a bug.

hakuturu583, 2023-12-21:
#1139 was merged

closed 2023-12-21 by hakuturu583