tier4/scenario_simulator_v2 #1122: getPolygon with 0 points

TauTheLepton, 2023-10-31:
**Describe the bug**
The `CatmullRomSpline` member function `getPolygon` when called with `num_points = 0` raises an error with a description `failed to calculate curve index`.
This description does not say where the real error is in this case. It is thrown deep from the call stack by member function `getCurveIndexAndS` in the call order presented below:
`getPolygon -> getLeftBounds -> getNormalVector -> getCurveIndexAndS`
This happens because the `num_points` (= 0) is used in `getLeftBounds` to calculate `step_size` (which is then equal to infinity).
https://github.com/tier4/scenario_simulator_v2/blob/c4b7daa714f2d7b94d6a71c0f78932b0826b4ca3/common/math/geometry/src/spline/catmull_rom_spline.cpp#L81
This leads to a NaN `s` value which then is not matched to any component curve and a mentioned exception is being thrown by `getCurveIndexAndS`.
https://github.com/tier4/scenario_simulator_v2/blob/c4b7daa714f2d7b94d6a71c0f78932b0826b4ca3/common/math/geometry/src/spline/catmull_rom_spline.cpp#L83
In my opinion there are three ways to improve the behavior in this situation:
- To catch the exception in the `getPolygon` function and thrown another exception with more informative description
- To just throw an exception in `getPolygon` when `num_points == 0`
- To add an early return in `getPolygon` to return an empty vector when `num_points == 0`

_Context:_
I am adding unit tests for the geometry package and wanted to add tests for the `getPolygon` function.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit any existing source code file or create a new one
2. Create any `CatmullRomSpline`
3. Call the `getPolygon` member function with -num_points` argument set to 0
4. Compile and execute
5. See the error

**Expected behavior**
Either an exception with an appropriate description or the return of an empty vector.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]: Ubuntu
 - Browser [e.g. chrome, safari]: Firefox
 - Version [e.g. 22]: 22
 - ROS 2 version: Humble
 - DDS: CycloneDDS

hakuturu583, 2023-11-01:
I think CatmullRomSpline class cannot construct with 0 control points.

https://github.com/tier4/scenario_simulator_v2/blob/c4b7daa714f2d7b94d6a71c0f78932b0826b4ca3/common/math/geometry/src/spline/catmull_rom_spline.cpp#L129-L133

So, I think it would not be a problem.
How did you test this issue?

TauTheLepton, 2023-11-02:
I am sorry if my explanation was not clear enough.
You are right, the `CatmullRomSpline` cannot be constructed with 0 control points.
However, the member function `getPolygon`
https://github.com/tier4/scenario_simulator_v2/blob/c4b7daa714f2d7b94d6a71c0f78932b0826b4ca3/common/math/geometry/src/spline/catmull_rom_spline.cpp#L30-L50
takes `num_points` as an argument. This argument is used to construct a polygon (or rather pairs of triangles) approximating the surface of the spline with a given width. It is independent from the number of control points that were used to construct the `CatmullRomSpline`.
This `num_points` argument is not checked anywhere, and when it is set to 0, then the behavior I have described above is triggered (which leads to an exception).

TauTheLepton, 2023-11-14:
@hakuturu583
Does the explanation in my previous message clearly explain where the bug is?

hakuturu583, 2023-11-24:
Sorry for my late reply.
I understand...
It is a bug.

TauTheLepton, 2023-12-21:
This issue was solved under #1139, but I forgot to mention this issue in the description of the #1139.
The #1139 was merged, so I am closing this issue.

closed 2023-12-21 by TauTheLepton