tier4/scenario_simulator_v2 #1123: CatmullRomSubspline collision not being detected

TauTheLepton, 2023-10-31:
**Describe the bug**
The `CatmullRomSubspline` member function `getCollisionPointIn2D` does not always return the correct result. In a simple case with one collision is works fine, but in more complicated scenarios with more collision points the function does not always return a collision point.
This is caused by the fact, that the `CatmullRomSubspline` calls the `CatmullRomSpline` member function `getCollisionPointIn2D` here:
https://github.com/tier4/scenario_simulator_v2/blob/c4b7daa714f2d7b94d6a71c0f78932b0826b4ca3/common/math/geometry/src/spline/catmull_rom_subspline.cpp#L28
which already filters out one collision point here (the first one found is returned):
https://github.com/tier4/scenario_simulator_v2/blob/c4b7daa714f2d7b94d6a71c0f78932b0826b4ca3/common/math/geometry/src/spline/catmull_rom_spline.cpp#L298-L320
This leads to incorrect situations, where a polygon has multiple collisions with the spline, the spline returns first (or last - depending on the `search_backward` argument) collision point, but this collision point is not part of the subspline. Some other collision point may be a part of the subspline, but this case is not checked, the information is lost and no collision point is returned from the `CatmullRomSubspline` member function `getCollisionPointIn2D`.

In my opinion there should exist a `CatmullRomSpline` member function that will return all collision points, so that from those collision points `CatmullRomSubspline` could choose the ones that are part of the spline and then return the first one (or the last one).
**NOTE** that this fix will require to modify the code that is affected by another bug (https://github.com/tier4/scenario_simulator_v2/issues/1120) which should probably be fixed first.

_Context:_
I am adding unit tests for the `geometry` package and wanted to add tests for the `getCollisionPointIn2D` function, but it generates the wrong results in some cases (see **To Reproduce**).

**To Reproduce**
I have created an example in the test case to demonstrate the bug [here](https://github.com/TauTheLepton/scenario_simulator_v2/blob/a781708f099297cbdb0a600cd14bb1eb3210ea2c/common/math/geometry/test/test_catmull_rom_spline.cpp#L357-L396)
Steps to reproduce the behavior:
1. Clone [this branch](https://github.com/TauTheLepton/scenario_simulator_v2/tree/example/subspline-collision-bug)
2. Build
3. Run tests
4. See that the last spline test passes (which makes no sens, as the same spline with the same polygon does collide in one case and does not collide in other case)

**Expected behavior**
The collision should always be detected (provided it exists).

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]: ubuntu
 - Browser [e.g. chrome, safari]: Firefox
 - Version [e.g. 22]: 22
 - ROS 2 version: Humble
 - DDS: CycloneDDS

hakuturu583, 2023-11-01:
I understand, it is a bug.

hakuturu583, 2023-12-21:
#1139 was merged

closed 2023-12-21 by hakuturu583