tier4/scenario_simulator_v2 #1161: Failure of scenario_test_runner

ZiweiSong96, 2023-12-22:
I installed scenario simulator v2 and autoware.universe on Ubuntu 22.04 with ROS2 humble. 
All the installation and build procedures are successful. 
When I started the scenario_test_runner. It seems failed: no Autoware and RVIZ2 appeared. 
Only the logging info kept showing up and I cannot get the results as shown in the instruction video or docs.
[image]

hakuturu583, 2023-12-25:
@ZiweiSong96 
It appears that the communication that should take place after launching Autoware is not taking place.
If possible, could you please provide error logs for other parts of the system?

ZiweiSong96, 2023-12-27:
It seems the simulator stucked at somewhere, I can only see the node or service "not ready". Besides, an information "Shutting down Autoware" appeared.
[image]
[image]

ZiweiSong96, 2023-12-27:
Actually these are all the error logs. They repeat one by one

hakuturu583, 2023-12-28:
It seems, you are trying to follow this instruction is it right?
https://tier4.github.io/AWSIM/GettingStarted/UsingOpenSCENARIO/

hakuturu583, 2023-12-28:
@mackierx111 Do you know something about this problem?

Lkaho, 2024-05-15:
Hi, @ZiweiSong96 , may I ask do you have any solution for this issue? I have the same problem.

jimbo-lin, 2024-06-04:
Hi, @Lkaho @ZiweiSong96, sorry to bother but i encounter same error, did you guys find a way to solve this problem?

ZiweiSong96, 2024-06-04:
> Hi, @ZiweiSong96 , may I ask do you have any solution for this issue? I have the same problem.

Sure, I still work on Autoware related project.

ZiweiSong96, 2024-06-04:
> Hi, @Lkaho @ZiweiSong96, sorry to bother but i encounter same error, did you guys find a way to solve this problem?

Actually I did not solve this question, so I turned to AWSIM and try to make some adjustment on the Unity 3D project to make some complex traffic scenarios.