tier4/scenario_simulator_v2 #1176: Can we get a LanePosition Publisher or something when the simulator run? I really need a simple enviroment

Zoe-Wan, 2024-01-24:
**Describe the solution you'd like**
I noticed that in the simulator, the car's position is represented by LanePosition at the beginning. I am curious if it is possible to continuously obtain the car's LanePosition (instead of WorldPosition) during the simulation? I really need this feature.

**Describe alternatives you've considered**
I think it is best to implement a laneposition publisher, but it is also nice to use API calls.

hakuturu583, 2024-01-24:
This information includes "entity/status" topic.
This topic publish [here.](https://github.com/tier4/scenario_simulator_v2/blob/6ba8e7819a39ceda5d0ff86c94b37546fba80821/simulation/traffic_simulator/src/entity/entity_manager.cpp#L780-L798)

Zoe-Wan, 2024-01-25:
Thanks, I'll check the code, and report the result here

hakuturu583, 2024-02-14:
I think this issue is no-longer active, please re-open issue if related problem happens.

closed 2024-02-14 by hakuturu583

Zoe-Wan, 2024-03-19:
Hi, me again. Thanks for giving help!
I'm using dummy_perception_publisher to test how noise can affect the car. To simply the situation I need create dummy object by using lanelet position, which seems not to be implemted.(?) Is there any other way to create perception noise on autoware siumulator? @hakuturu583

Zoe-Wan, 2024-03-19:
> Hi, me again. Thanks for giving help! I'm using dummy_perception_publisher to test how noise can affect the car. To simply the situation I need create dummy object by using lanelet position, which seems not to be implemted.(?) And is there any other way to create perception noise (such as fake car, fake pointcloud) on autoware siumulator? @hakuturu583
Opps, I find that in scenario_simulator I should use **simple sensor simulator** but not **dummy perception publisher** 😢 . But unfortunately I still have no idea how to make it... I use python language but the **simple sensor simulator** package is implemented by cpp. Can you give an example about how to use this package?

Zoe-Wan, 2024-03-20:
I reimplement a new sensor simulator, but how can I remove the old one? ...need some help indeed..

Zoe-Wan, 2024-03-21:
OK，I find the way.