tier4/scenario_simulator_v2 #1179: Failure of recording a scenario simulation

lethal233, 2024-01-26:
**Describe the bug**
A clear and concise description of what the bug is.

The scenario simulator cannot perform normally when I set the `record` to true, which means:
1. it cannot do the scenario test simulation
2. it cannot record the simulation.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the scenario test simulation with the command `ros2 launch scenario_test_runner scenario_test_runner.launch.py   architecture_type:=awf/universe record:=false scenario:='t4v2.yml'   sensor_model:=sample_sensor_kit   vehicle_model:=sample_vehicle`
2. The scenario is simulated and played on the rviz.
3. Run the scenario test simulation again with the command `ros2 launch scenario_test_runner scenario_test_runner.launch.py   architecture_type:=awf/universe record:=true scenario:='t4v2.yml'   sensor_model:=sample_sensor_kit   vehicle_model:=sample_vehicle`
5. Rviz is successfully started, but the scenario is not simulated, along with console error output.

**Expected behavior**

The scenario test simulator should run the scenario normally and record the whole simulation messages.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - ROS 2 version: humble

The file is attached here: https://drive.google.com/file/d/1LDSyvA9PUEf58OzcS4sJ-3XLTVgAnOPJ/view?usp=drive_link

hakuturu583, 2024-01-30:
As we infer from the error text, it is probably an error in the DDS communication process.
It may be cured by reviewing the DDS communication settings, etc.

lethal233, 2024-01-30:
Got it, I will try to inspect the DDS settings first.

lethal233, 2024-01-30:
@hakuturu583 Thanks, after configuring the DDS settings I can record and replay the scenario mentioned above successfully. But there is another issue, when I replay the rosbag file, the steering angle and speed were not displayed. I also tried the sample file from https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/, and there was also no such information. Screenshot is attached.
[image]

closed 2024-01-30 by lethal233