tier4/scenario_simulator_v2 #1202: Compilation of openscenario_interpreter fails

ralwing, 2024-02-28:
**Describe the bug**
Building current `autoware.universe` with scenario_simmulator fails:
```
Finished <<< bluetooth_monitor [2.32s]
--- stderr: openscenario_interpreter                                                                                                                                                                                                                         
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/openscenario_interpreter.dir/build.make:132: CMakeFiles/openscenario_interpreter.dir/src/syntax/acquire_position_action.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/openscenario_interpreter.dir/build.make:188: CMakeFiles/openscenario_interpreter.dir/src/syntax/add_entity_action.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/openscenario_interpreter.dir/build.make:1070: CMakeFiles/openscenario_interpreter.dir/src/syntax/lane_change_action.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/openscenario_interpreter.dir/build.make:426: CMakeFiles/openscenario_interpreter.dir/src/syntax/condition.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/openscenario_interpreter.dir/build.make:160: CMakeFiles/openscenario_interpreter.dir/src/syntax/action.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:159: CMakeFiles/openscenario_interpreter.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< openscenario_interpreter [1min 21s, exited with code 2]
```
 

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the Dev docker installation steps: https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation-devel/
2. Add driving_log_replayer: https://tier4.github.io/driving_log_replayer/quick_start/installation/
3. Compile: `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS=ON`
4. Result:
```
Summary: 390 packages finished [2min 23s]
  1 package failed: openscenario_interpreter
  4 packages had stderr output: autoware_launch driving_log_replayer openscenario_interpreter ros2bag_extensions
  4 packages not processed

```

**Expected behavior**
All packages are buildable.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Ubuntu: 22.04
 - ROS 2 version
```
#env  
ROS_VERSION=2
NVIDIA_VISIBLE_DEVICES=all
ROS_PYTHON_VERSION=3
HOSTNAME=c034a2abebfe
PWD=/home/gigi/workspace/autoware
NVIDIA_DRIVER_CAPABILITIES=all
XAUTHORITY=/tmp/.dockerv737a0k9.xauth
HOME=/home/gigi
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
AMENT_PREFIX_PATH=/opt/ros/humble
PYTHONPATH=/opt/ros/humble/lib/python3.10/site-packages:/opt/ros/humble/local/lib/python3.10/dist-packages
TERM=xterm-256color
SHLVL=1
LD_LIBRARY_PATH=/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/x86_64-linux-gnu:/opt/ros/humble/lib
ROS_LOCALHOST_ONLY=0
PATH=/opt/ros/humble/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ROS_DISTRO=humble
QT_X11_NO_MITSHM=1
OLDPWD=/home/gigi
_=/usr/bin/env
```

HansRobo, 2024-02-28:
@ralwing Thank you for reporting!

However, this error is something that we as developers and our daily build CI rarely encounter.
I don't think it's a problem with the source code, it's just that there isn't enough memory when compiling.
Could you please reduce the number of parallel builds to reduce memory consumption or increase swap to increase the amount of usable memory and try again?
The instructions are here.
https://autowarefoundation.github.io/autoware-documentation/main/support/troubleshooting/#build-issues

ralwing, 2024-02-29:
@HansRobo Thank You. Following these instructions fixed the problem. I had to customize them a bit, but finally made it:
on my host:
I had to create swap file on my host instead of the docker:
```
sudo dd if=/dev/zero of=/swapfile bs=1G count=32
sudo chmod 666 /swapfile
sudo mkswap /swapfile
sudo swapon /swapfile
```
and than run docker with privileged rights and swapfile volume
```
/usr/bin/python3 /usr/bin/rocker --nvidia --x11 --user --volume /home/gigi/workspace/autoware --volume /home/gigi/workspace/autoware_map --volume /swapfile:/swapfile:rw --privileged -- ghcr.io/autowarefoundation/autoware-universe:latest-cuda
``` 

Thanks anyway :-)

closed 2024-02-29 by ralwing