tier4/scenario_simulator_v2 #1215: We need an API for python!!

Zoe-Wan, 2024-03-21:
**Is your feature request related to a problem? Please describe.**
I need an information about whether my ego vehicle is collision every frame, but unfortunately the checkCollision function is coded in cpp, and I can't use it. I'm trying to reimplement it in python, but it really cost a lot. :cry: 

**Describe the solution you'd like**
As you know, the autoware support both cpp and python, and I think there are a lot of people want to use your simulator as an reinforcement enviroment as I do. So, I think we need an API of traffic simulator for python :+1: 
( It's better if you can implement a simple sensor simulator in python :laughing: , mine is quite simple.

hakuturu583, 2024-03-22:
Thanks for the idea! It would certainly be useful to have a Python API.
However, as discussed within the TIER IV simulation team, we don't have the capacity to support a Python API. Sorry.
I think it would be possible to create an API wrapper for traffic_simulator using something like PyBind11 and control the scenario with Python.
If you could contribute, please do!

Zoe-Wan, 2024-03-22:
Ah! Sorry to hear that . It's a big work, I don't think I can handle it by my own... :cry:, I'll try your suggestion!

Zoe-Wan, 2024-03-22:
And there is another problem... I'm confusing about how do the 'DetectedObjects' ( published on /perception/object_recognition/detection/objects, can provided by simple sensor simulator) translate to the 'PredictedObjects' ( published on /perception/object_recognition/objects ). I find that PredictedObjects have more attributes than DetectedObjects ( like waypoint or something ), but I can't create it by myself. What should I do if I want to disturb the planning module of the Autoware?(You know, they usually use PredictedObjects for Motion Planning/Avoiding. )

closed 2024-04-18 by hakuturu583