tier4/scenario_simulator_v2 #1221: Simulator automatically limit the NPC velocity if it is behind of EGO

brkay54, 2024-04-02:
**Describe the bug**

- I am trying to create a scenario, I am sharing the details below:

[image]
```
**Rear-to-Front Vehicle Crossover**

**Initial condition**

- EGO is running at Ve. There is a motorcycle behind the EGO at Vn.

**Action** 

- A motorcycle comes from the rear, overtakes EGO, and integrates into EGO's forward trajectory, positioning itself directly in front of EGO.

**Success condition**

- EGO does not sudden brake
- EGO reaches the destination

**Failure condition**

- Collide with NPC

**Variables**

- Ve: [30, 20, 10] km/h
- Vn: Ve * 1.2 km/h
```

- When I created the scenario, the NPC can not overtake the EGO because the simulator is limitting the NPC velocity (Iff they are in the same lane.).

[!limited-vel](https://gist.github.com/assets/45468306/f8ebf786-a9d7-4a50-8978-e01433e27001)

- I tried some other methods but the behavior was same. 

**To Reproduce**
- I added map and scenario file below:
[overtake-scenario.tar.gz](https://github.com/tier4/scenario_simulator_v2/files/14836839/overtake-scenario.tar.gz)

**Expected behavior**
- Velocity of the NPC is higher than EGO in scenario definition so NPC should be able to overtake the EGO.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Ubuntu 22.04
 - ROS 2 Humble
 - CycloneDDS

hakuturu583, 2024-04-04:
> Although I set isBLIND: True, NPC can not overtake the EGO because the simulator is limitting the NPC velocity (Iff they are in the same lane.).

This behavior is as per scenario_simulator specifications.
The behavior_plugin currently used by default decelerates the Entity in front of it in the lane coordinate system.

Currently, I suggest to write scenarios in 2 ways.

1. Use followTrajectoryAction
2. Add left lane and use laneChangeAction.

hakuturu583, 2024-04-04:
In 2nd way, adding left lane and put the motorcycle on the left lane and start.
[image]

hakuturu583, 2024-04-12:
I think this issue is not active.
If you still working on this issue, please re-open this issue.

closed 2024-04-12 by hakuturu583

brkay54, 2024-04-15:
@hakuturu583 Sorry, I couldn't reply. I was working on to define npc behavior with your nicely tips. Thank you for the answer. If I can not create the scenario, I will re-open. 🙏🏼