tier4/scenario_simulator_v2 #1236: How to specify the state of traffic lights in the scenario?

hcn1519, 2024-04-22:
I am running a scenario using the scenario test runner, and I wonder how to specify the state of traffic lights in the [TIER IV Scenario Format Version 2.0](https://tier4.github.io/scenario_simulator_v2-docs/developer_guide/TIERIVScenarioFormatVersion2/). It seems that the scenario simulator v2 does not follow the [OpenScenario v1.2 specification](https://publications.pages.asam.net/standards/ASAM_OpenSCENARIO/ASAM_OpenSCENARIO_XML/v1.2.0/10_scenario_creation/10_10_traffic_signal.html).

Below are my scenario and map files, which I used to test traffic light behaviors. The scenario includes a simple ego routing story and sets all traffic lights to red. (I have checked the syntax against [other scenario files](https://github.com/tier4/scenario_simulator_v2/blob/2dbab8a177ea0b02323f8022f60c276238536bc7/test_runner/scenario_test_runner/scenario/CustomCommandAction.V2ITrafficSignalStateAction.yaml), though there might be an error.) I expected the ego car to stop at the junction because of the traffic light, but it ignores it and reaches the destination.

- [test_scenario_map.zip](https://github.com/tier4/scenario_simulator_v2/files/15056119/test_scenario_map.zip)

The code below is the part related to traffic lights in the scenario file.

```yaml
RoadNetwork:
  LogicFile:
    filepath: lanelet2_map.osm
  SceneGraphFile:
    filepath: point_cloud.pcd
  TrafficSignals:
      TrafficSignalController:
      - name: test_signals
        Phase:
        - duration: INF
          name: Stop
          TrafficSignalState:
          - state: red solidOn circle
            trafficSignalId: '494'
          - state: red solidOn circle
            trafficSignalId: '520'
          - state: red solidOn circle
            trafficSignalId: '586'
          - state: red solidOn circle
            trafficSignalId: '553'
Storyboard:
    Init:
      Actions:
        GlobalAction:
        - InfrastructureAction:
            TrafficSignalAction:
              TrafficSignalControllerAction:
                phase: Stop
                trafficSignalControllerRef: test_signals
```

HansRobo, 2024-04-22:
@hcn1519 

Hi, thank you for reporting!

Since the traffic light message type differs depending on the version of autoware, it is necessary to appropriately switch the traffic light topic output from the scenario_simulator_v2 side.

You can switch it by switching the `architecture_type` of scenario_test_runner.
For older Autoware, `awf/universe` is appropriate, and for newer Autoware, `awf/universe/20230906` is appropriate.

If nothing is specified, `awf/universe` will be used, so the traffic light will not connect to new autoware.

We are aware of this issue and are working with #1232 .

HansRobo, 2024-04-22:
> It seems that the scenario simulator v2 does not follow the [OpenScenario v1.2 specification](https://publications.pages.asam.net/standards/ASAM_OpenSCENARIO/ASAM_OpenSCENARIO_XML/v1.2.0/10_scenario_creation/10_10_traffic_signal.html).

The user guide provides a scenario for specifying a traffic light using OpenDRIVE as an example.
However, in reality, as shown below, this is just an example, and it is not stipulated in the standard that it must be followed.

[description of TrafficSignalState::state](https://publications.pages.asam.net/standards/ASAM_OpenSCENARIO/ASAM_OpenSCENARIO_XML/v1.2.0/generated/content/TrafficSignalState.html#:~:text=XSDattribute-,State%20of%20the%20signal%2C%20e.g.%20the%20visual%20information%20%22off%3Boff%3Bon%22.%20The%20number%20of%20values%20in%20the%20state%20depends%20on%20the%20traffic%20signal%20in%20the%20roadnetwork%20%2C%20e.g.%20type%20and%20subtype%20in%20OpenDRIVE,-trafficSignalId)

> State of the signal, e.g. the visual information "off;off;on". The number of values in the state depends on the traffic signal in the roadnetwork , e.g. type and subtype in OpenDRIVE

Autoware uses lanelet2 as its map format instead of OpenDRIVE, so please understand that although it is within the scope of the OpenSCENARIO standard, the specifications are different from the scenario in the user guide, which assumes OpenDRIVE.

hcn1519, 2024-04-22:
Thank you for your answer @HansRobo 
I test to use `awf/universe/20230906`, and it works. Also, I appreciate your additional answer regarding specification.

HansRobo, 2024-11-19:
If you are still having problems please reopen this or create a new issue.

closed 2024-11-19 by HansRobo