tier4/scenario_simulator_v2 #1240: Hsinchu city (Taiwan) Map Bug - Lane=1720 is unreachable from any other lane on the map

lethal233, 2024-05-02:
**Describe the bug**

When I place the ego car on lane=1718 on the Hsinchu city (Taiwan) map and set the destination point to somewhere on lane=1720 , the ego car will not move during the simulation. And the simulation task failed.

[image]

Please see the TW map below, intuitively, according to the map visualization, the lane=1720 and lane=1721 should be reachable from lane=1718 (lane=1720 and lane=1721 are successors of lane=1718). 

[image]

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Launch Autoware through docker
2. Run the scenario `1720.yml` (at the end of this report), which uses `Hsinchu city (Taiwan)` map. I downloaded it from [here](https://evaluation.tier4.jp/evaluation/maps)
```sh
ros2 launch scenario_test_runner scenario_test_runner.launch.py \
    architecture_type:=awf/universe/20230906 \
    record:=false \
    scenario:=1720.yml \
    sensor_model:=sample_sensor_kit \
    vehicle_model:=sample_vehicle
```

3. Simulation runtime log:
```sh
[openscenario_interpreter_node-3] [INFO] [1714688055.375044352] [simulation.openscenario_interpreter]: InternalError: Requested the service "/api/routing/set_route_points" 1 times, but was unsuccessful.
[scenario_test_runner.py-1] [INFO] [1714688055.379492981] [simulation.scenario_test_runner]: finish execution
```
4. See the simulation result log:
```
<?xml version="1.0"?>
<testsuites name="/tmp/scenario_test_runner" failures="0" errors="1" tests="1">
  <testsuite name="1720" failures="0" errors="1" tests="1">
    <testcase name="1720_0">
      <error type="InternalError" message="Requested the service &quot;/api/routing/set_route_points&quot; 1 times, but was not successful." />
    </testcase>
  </testsuite>
</testsuites>
```

**Expected behavior**
Expect routing to find valid route from lane=1718 to lane=1720

**Root Cause**
I think the reason behind this bug is:
1. The end point of lane=1718's left bound != the start point of lane=1720's left bound
2. The end point of lane=1718's right bound != the start point of lane=1720's right bound

I use your lanelet2 extension python library to get this:
```
lane=1718 left bound:  [id: 1685, inverted point ids: 1688, 1687, 1686, 1684, 1683, 1708, 1699]
lane=1718 right bound:  [id: 1679 point ids: 1677, 1678, 1680, 1681, 1717, 1694]
lane=1720 left bound:  [id: 1704, inverted point ids: 1708, 1707, 1706, 1705, 1703, 1702, 1656]
lane=1720 right bound:  [id: 1711, inverted point ids: 1717, 1716, 1715, 1714, 1713, 1712, 1710, 1709, 1658]
```

I tried to add point=1699 to lane=1720's left bound, and point=1694 to lane=1720's right bound. The modified version is like this:

[image]

And I did the simulation again, it worked. The ego car can depart from lane=1718 and arrive at lane=1720:

[image]

[image]

**Desktop (please complete the following information):**
 - Ubuntu 22.04
[... 196 lines cut ...]
                                  delay: 0
                                  conditionEdge: none
                                  ByValueCondition:
                                    SimulationTimeCondition:
                                      value: 180
                                      rule: greaterThan
                        Action:
                          - name: ''
                            UserDefinedAction:
                              CustomCommandAction:
                                type: exitFailure
            StartTrigger:
              ConditionGroup:
                - Condition:
                    - name: ''
                      delay: 0
                      conditionEdge: none
                      ByValueCondition:
                        SimulationTimeCondition:
                          value: 0
                          rule: greaterThan
    StopTrigger:
      ConditionGroup: []

```

simonranjith, 2025-04-04:
Hello there, I'm facing a similar issue with a map. How do we use the lanelet2 extension python library? Can someone guide me? Thanks.

hakuturu583, 2025-04-07:
@simonranjith Sorry for our late reply.
Could you share the map you found problem?

simonranjith, 2025-04-07:
@hakuturu583 I have pasted my map below. I would like the ego to move on laneId='109'. But I get the following error message:  "Requested the service "/api/routing/set_route_points" 1 times, but was not successful." Do you have any suggestions on how to debug this issue? Thanks for your help.

<?xml version="1.0" encoding="UTF-8"?>
<osm generator="VMB">
  <MetaInfo format_version="1" map_version="5" validation_version=""/>
  <node id="1" lat="40.63394105284" lon="-8.62994614974">
    <tag k="local_x" v="-1.6194"/>
    <tag k="local_y" v="186.2679"/>
    <tag k="ele" v="-1.3995"/>
  </node>
  <node id="2" lat="40.63306761338" lon="-8.63093334292">
    <tag k="local_x" v="-85.1026"/>
    <tag k="local_y" v="89.3145"/>
    <tag k="ele" v="-2.6292"/>
  </node>
  <node id="7" lat="40.63393288545" lon="-8.62993216762">
    <tag k="local_x" v="-0.437"/>
    <tag k="local_y" v="185.3613"/>
    <tag k="ele" v="-1.4122"/>
  </node>
  <node id="8" lat="40.63305907856" lon="-8.63091828477">
    <tag k="local_x" v="-83.8292"/>
    <tag k="local_y" v="88.3671"/>
    <tag k="ele" v="-2.5959"/>
  </node>
  <node id="10" lat="40.63391526692" lon="-8.62990526882">
    <tag k="local_x" v="1.8377"/>
    <tag k="local_y" v="183.4056"/>
    <tag k="ele" v="-1.4064"/>
  </node>
  <node id="11" lat="40.63304146025" lon="-8.63089138606">
    <tag k="local_x" v="-81.5545"/>
    <tag k="local_y" v="86.4114"/>
    <tag k="ele" v="-2.5671"/>
  </node>
  <node id="14" lat="40.63389764837" lon="-8.62987837004">
    <tag k="local_x" v="4.1124"/>
    <tag k="local_y" v="181.4499"/>
    <tag k="ele" v="-1.4105"/>
  </node>
  <node id="15" lat="40.63302384194" lon="-8.63086448738">
    <tag k="local_x" v="-79.2798"/>
    <tag k="local_y" v="84.4557"/>
    <tag k="ele" v="-2.5584"/>
  </node>
  <node id="18" lat="40.63388860352" lon="-8.62986457832">
    <tag k="local_x" v="5.2787"/>
    <tag k="local_y" v="180.4459"/>
    <tag k="ele" v="-1.3995"/>
  </node>
  <node id="19" lat="40.63301516475" lon="-8.63085177178">
    <tag k="local_x" v="-78.2045"/>
    <tag k="local_y" v="83.4925"/>
    <tag k="ele" v="-2.6292"/>
  </node>
  <node id="21" lat="40.63401709893" lon="-8.629907298">
    <tag k="local_x" v="1.6661"/>
    <tag k="local_y" v="194.7092"/>
    <tag k="ele" v="-1.5039"/>
  </node>
  <node id="22" lat="40.63417414034" lon="-8.63014954056">
    <tag k="local_x" v="-18.8191"/>
    <tag k="local_y" v="212.1412"/>
    <tag k="ele" v="-2.9087"/>
  </node>
  <node id="24" lat="40.63399654623" lon="-8.62993025548">
    <tag k="local_x" v="-0.2753"/>
    <tag k="local_y" v="192.4278"/>
    <tag k="ele" v="-1.4873"/>
  </node>
[... 6263 lines cut ...]
    <nd ref="1418"/>
    <nd ref="1419"/>
    <nd ref="1420"/>
    <nd ref="1421"/>
    <nd ref="1422"/>
    <nd ref="1423"/>
    <nd ref="21"/>
    <tag k="type" v="virtual"/>
  </way>
  <relation id="13">
    <member type="way" role="left" ref="9"/>
    <member type="way" role="right" ref="12"/>
    <tag k="type" v="lanelet"/>
    <tag k="subtype" v="road"/>
    <tag k="speed_limit" v="50"/>
    <tag k="location" v="urban"/>
    <tag k="one_way" v="yes"/>
  </relation>
  <relation id="17">
    <member type="way" role="left" ref="12"/>
    <member type="way" role="right" ref="16"/>
    <tag k="type" v="lanelet"/>
    <tag k="subtype" v="road"/>
    <tag k="speed_limit" v="50"/>
    <tag k="location" v="urban"/>