tier4/scenario_simulator_v2 #1260: 'scenario_test_runner' doesn't work with Autoware

halilsenaydin, 2024-05-21:
**Describe the bug**
I installed scenario simulator v2 and autoware.universe on Ubuntu 22.04 with ROS2 humble.
All the installation and build procedures are successful.
When I started the scenario_test_runner. It seems failed: 'AutowareError: Simulator waited for the Autoware state to transition to WaitingForRoute, but time is up. The current Autoware state is INITIALIZING.'

**Full log**
[run_3.txt](https://github.com/tier4/scenario_simulator_v2/files/15391152/run_3.txt)
[run_2.txt](https://github.com/tier4/scenario_simulator_v2/files/15391153/run_2.txt)
[run_1.txt](https://github.com/tier4/scenario_simulator_v2/files/15391154/run_1.txt)

**Expected behavior**
I'm waiting for the scenario simulation to run with Autoware.

**Screenshots**
# scenario:='/home/arge01/Downloads/scenerios/lanelet2_map/success/crosswalk_v2.yaml
[image]
[image]

**Desktop (please complete the following information):**
 - Ubuntu Version 22.04
 - ROS 2 Version Humble
 - Cyclone DDS

hakuturu583, 2024-05-22:
@halilsenaydin Could you provide the version of Autoware you used?

halilsenaydin, 2024-05-23:
Branch: release/2024.04

halilsenaydin, 2024-05-23:
> @halilsenaydin Could you provide the version of Autoware you used?

@hakuturu583  I am using,

Branch: release/2024.04

hakuturu583, 2024-05-24:
Could you try with

```
ros2 launch scenario_test_runner scenario_test_runner.launch.py architecture_type:=awf/universe/20230906 record:=false scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml' sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle
```

halilsenaydin, 2024-05-25:
Unfortunately this doesn't work. I am started it the Autoware workspace and the script _v2 workspace. Both return errors.

[bug-1.txt](https://github.com/tier4/scenario_simulator_v2/files/15444909/bug-1.txt)
[bug-2.txt](https://github.com/tier4/scenario_simulator_v2/files/15444910/bug-2.txt)

hakuturu583, 2024-05-27:
```
[openscenario_interpreter_node-3] [WARN] [1716667180.184187593] [simulation.openscenario_interpreter]: Your machine is not powerful enough to run the scenario at the specified frame rate (30 Hz). We recommend that you reduce the frame rate to 4.52489 or less.
```

In this log, warnings like this are remians.
This warning means your computer is not powerful enough.
Could you try with more powerful computer?

closed 2024-05-27 by halilsenaydin