tier4/scenario_simulator_v2 #1272: Scenario simulator does not print or visualize parameters/modifier used for test iterations

ahmeddesokyebrahim, 2024-06-04:
**Is your feature request related to a problem? Please describe.**
While debugging scenarios with different values (list) for modifiers, it is not clear which values is selected and used between different combinations in a specific test iteration.

For example, a scenario with modifiers like that ...
```
    - name: __tier4_modifier_an
      list:
        - 0.2
        - 2
    - name: __tier4_modifier_D
      list:
        - 5
        - 10
```  
it becomes confusing which values of `__tier4_modifier_an` is used with which value of ` __tier4_modifier_D` in an iteration

**Describe the solution you'd like**
It will be helpful if the parameters and modifiers used for each iteration is visualized in RViz.

**Describe alternatives you've considered**
It could a valid alternative to have it only printed in the terminal instead of text visualization in RViz

**Additional context**

mitsudome-r, 2024-08-13:
@yamacir-kit Do you have any opinion about how these information should be output to the user? (in Rviz, in terminal, make the verbosity configurable, etc.)

yamacir-kit, 2026-11-12:
I am closing this issue together with #1360, which was closed because it had become too old to merge.

If this feature is still needed, please open a new issue or a new pull request based on the current `develop` branch.

closed 2026-11-12 by yamacir-kit