tier4/scenario_simulator_v2 #1302: The position of kinematic_state topic is not updated

brkay54, 2024-06-25:
**Describe the bug**
When I run the scenario simulator, the position of the ego stays in the initial position:

https://github.com/tier4/scenario_simulator_v2/assets/45468306/31b1f26f-2248-4840-ae32-a03680ba65a3

As you can see in the video, although the velocity is increasing, the ego stays in the same position. 
When I checked the `/localization/kinematic_state`, the velocity message was changing but the position info stayed the same value.

I also run Autoware without scenario_simulator, I didn't see this problem. 

**To Reproduce**
Steps to reproduce the behavior:
- Run scenario simulator

**Expected behavior**
The position of the ego should be updated.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Ubuntu Version: 22.04
 - ROS 2 version: Humble
 - DDS: cyclonedds

kyoichi-sugahara, 2024-06-25:
@brkay54 
Hello, 
Although I'm not very knowledgeable about scenario simulators, I sometimes waste time by not noticing simple mistakes, so I'd like to comment.
I'd like to try reproducing this on my environment. Would it be possible for you to share the files and maps used during the simulator execution?
At the very least, I haven't encountered this issue when using a different scenario and map in my own environment.
Also, if possible, I'd like to see the logs output to the terminal.

Additionally, with the default RViz settings, the control outputs should be visualized, but it seems they're not included, which concerns me a bit.

brkay54, 2024-06-25:
@kyoichi-sugahara -san, thanks for your reply, 
> Although I'm not very knowledgeable about scenario simulators, I sometimes waste time by not noticing simple mistakes, so I'd like to comment.

I totally understand your concerns and feelings about the simple mistakes 🥲 

[sample-scenario.tar.gz](https://github.com/user-attachments/files/15974987/sample-scenario.tar.gz)

I added the map and scenario file.

> At the very least, I haven't encountered this issue when using a different scenario and map in my own environment.

I tested with different scenario and map file too, but the same problem occurred.

Here is my terminal log:

<details>
<summary>Click to expand</summary>

```bash
berkay@victus:~/projects/autoware$ ros2 launch scenario_test_runner scenario_test_runner.launch.py    record:=false   scenario:='/home/berkay/projects/awf-evaluation-scenarios/scenarios/Urban_Scenarios/tutorial-scenario.yaml'   sensor_model:=sample_sensor_kit   vehicle_model:=sample_vehicle
[INFO] [launch]: All log files can be found below /home/berkay/.ros/log/2024-06-25-20-16-43-270830-victus-393837
[INFO] [launch]: Default logging verbosity is set to INFO
architecture_type                   := awf/universe/20240605
autoware_launch_file                := planning_simulator.launch.xml
autoware_launch_package             := autoware_launch
consider_acceleration_by_road_slope := False
consider_pose_by_road_slope         := True
enable_perf                         := False
global_frame_rate                   := 30.0
global_real_time_factor             := 1.0
global_timeout                      := 180
initialize_duration                 := 30
launch_autoware                     := True
launch_rviz                         := False
output_directory                    := /tmp
port                                := 5555
record                              := false
rviz_config                         := 
scenario                            := /home/berkay/projects/awf-evaluation-scenarios/scenarios/Urban_Scenarios/tutorial-scenario.yaml
sensor_model                        := sample_sensor_kit
sigterm_timeout                     := 8
use_sim_time                        := False
vehicle_model                       := sample_vehicle
[INFO] [scenario_test_runner.py-1]: process started with pid [393838]
[INFO] [simple_sensor_simulator_node-2]: process started with pid [393840]
[INFO] [openscenario_interpreter_node-3]: process started with pid [393842]
[INFO] [openscenario_preprocessor_node-4]: process started with pid [393844]
[INFO] [visualization_node-5]: process started with pid [393846]
[scenario_test_runner.py-1] [INFO 1719335803.733213172] [simulation.scenario_test_runner]: connection established with preprocessor (print_debug() at /home/berkay/projects/autoware/install/scenario_test_runner/lib/scenario_test_runner/scenario_test_runner.py:268)
[scenario_test_runner.py-1] [INFO 1719335805.001877205] [simulation.scenario_test_runner]: Result of /simulation/openscenario_preprocessor/load: success (print_debug() at /home/berkay/projects/autoware/install/scenario_test_runner/lib/scenario_test_runner/scenario_test_runner.py:268)
[scenario_test_runner.py-1] [INFO 1719335805.003416599] [simulation.scenario_test_runner]: Result of /simulation/openscenario_preprocessor/check: True (print_debug() at /home/berkay/projects/autoware/install/scenario_test_runner/lib/scenario_test_runner/scenario_test_runner.py:268)
[scenario_test_runner.py-1] [INFO 1719335805.004905183] [simulation.scenario_test_runner]: derived : /tmp/scenario_test_runner/tutorial-scenario/tutorial-scenario.xosc (print_debug() at /home/berkay/projects/autoware/install/scenario_test_runner/lib/scenario_test_runner/scenario_test_runner.py:268)
[scenario_test_runner.py-1] [INFO 1719335805.006232899] [simulation.scenario_test_runner]: Result of /simulation/openscenario_preprocessor/check: False (print_debug() at /home/berkay/projects/autoware/install/scenario_test_runner/lib/scenario_test_runner/scenario_test_runner.py:268)
[scenario_test_runner.py-1] [INFO 1719335805.007060121] [simulation.scenario_test_runner]: finish derivation (print_debug() at /home/berkay/projects/autoware/install/scenario_test_runner/lib/scenario_test_runner/scenario_test_runner.py:268)
[scenario_test_runner.py-1] [INFO 1719335805.007872517] [simulation.scenario_test_runner]: /tmp/scenario_test_runner/tutorial-scenario/tutorial-scenario.xosc (print_debug() at /home/berkay/projects/autoware/install/scenario_test_runner/lib/scenario_test_runner/scenario_test_runner.py:268)
[scenario_test_runner.py-1] [INFO 1719335805.008603584] [simulation.scenario_test_runner]: Run tutorial-scenario.xosc (1 of 1) (run_preprocessed_scenarios() at /home/berkay/projects/autoware/install/scenario_test_runner/lib/scenario_test_runner/scenario_test_runner.py:199)
[openscenario_interpreter_node-3] [INFO] [launch]: All log files can be found below /home/berkay/.ros/log/2024-06-25-20-16-47-497642-victus-394130
[openscenario_interpreter_node-3] [INFO] [launch]: Default logging verbosity is set to INFO
[simple_sensor_simulator_node-2] [WARN 1719335807.600356836] [tf2_buffer]: Detected time source change. Clearing TF buffer. (onTimeJump() at ./src/buffer.cpp:105)
[openscenario_interpreter_node-3] [INFO 1719335807.678291409] [simulation.concealer_user]: /api/autoware/set/velocity_limit service is not ready. (validateAvailability() at /home/berkay/projects/autoware/src/simulator/scenario_simulator/external/concealer/include/concealer/se [... cut ...] tion/share/sample_vehicle_description/config/simulator_model.param.yaml'].
```

</details>

> Additionally, with the default RViz settings, the control outputs should be visualized, but it seems they're not included, which concerns me a bit.

Actually, I was creating scenarios by using Scenario Simulator and I didn't update my Autoware work environment for a month. Today I pulled all the updates. By default, I am on `RJD-736/autoware_msgs_support` branch, and I didn't do any change on code base.

hakuturu583, 2024-07-02:
@brkay54 Could you provide branch or commit hash of the Autoware you used.

brkay54, 2024-07-02:
Hi, @hakuturu583 -san, thank you for your reply. This week I updated my workspace to the latest commit again. Today I tried to run the scenario simulator again but I couldn't play the same scenario because of the error: `[simulation.concealer_user]: Simulator waiting for Autoware state to be WaitingForRoute (remain: 211).`.

It is because of the mrm_handler initialized the vehicle with `EMERGENCY`, so I couldn't run. Created an issue about it:
- https://github.com/autowarefoundation/autoware.universe/issues/7792
For temporal solution I commented out the if condition [here](https://github.com/tier4/tier4_ad_api_adaptor/blob/tier4/universe/autoware_iv_internal_api_adaptor/src/iv_msgs.cpp#L54-L56). After disable it I can run the scenario_simulator, it can set route now.

However, the behavior was same, the kinematic_state does not update the position. I can see the velocity increasing but position is same. My commit sha in universe: ce099071716d57abb19d7eb10c6e0b4e65bbe0db

ahmeddesokyebrahim, 2024-07-03:
Thanks for opening this issue @brkay54 .
Regarding your comment @kyoichi-sugahara -san 

>  At the very least, I haven't encountered this issue when using a different scenario and map in my own environment.

I confirm having the same issue like @brkay54 and the following video shows the vehicle position is not updated while the `/localization/kinematic_state` topic is updated.

I am using a similar configuration like Berkay's 

<details>
<summary>click to expand</summary>

```
architecture_type                   := awf/universe/20240605
autoware_launch_file                := planning_simulator.launch.xml
autoware_launch_package             := autoware_launch
consider_acceleration_by_road_slope := False
consider_pose_by_road_slope         := True
enable_perf                         := False
global_frame_rate                   := 30.0
global_real_time_factor             := 1.0
global_timeout                      := 180
initialize_duration                 := 60
launch_autoware                     := True
launch_rviz                         := False
output_directory                    := /tmp
port                                := 5555
record                              := false
rviz_config                         := 
scenario                            := /workspace/scenarios/UC-NTR-003-0004_case1.yaml
sensor_model                        := sample_sensor_kit
sigterm_timeout                     := 8
use_sim_time                        := False
vehicle_model                       := sample_vehicle
```
</details>

Please let us know if you have any thoughts as this is considered as a blocker for Berkay currently and it will be a blocker for me very soon.

kyoichi-sugahara, 2024-07-03:
@brkay54 
> I added the map and scenario file.

So sorry I mised your mention :bow: 
I will try to reproduce the issue tomorrow. cc: @ahmeddesokyebrahim 

> It is because of the mrm_handler initialized the vehicle with EMERGENCY, so I couldn't run. Created an issue about it:

I faced the same problem and solved in the [PR](https://github.com/autowarefoundation/autoware.universe/pull/7784) and this is caused by the [change](https://github.com/autowarefoundation/autoware_launch/pull/1043) which is merged last Friday. So my pr is not the solution for this kinemtic_state issue I guess.

kyoichi-sugahara, 2024-07-03:
@brkay54 @ahmeddesokyebrahim 
By the way, does the simulator mentioned in the [tutorial](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/scenario-simulation/planning-simulation/scenario-test-simulation/#scenario-test-simulation) work in your environment?
I've only looked at the downloaded files, but I'm a bit concerned that in sample.yaml, there are definitions like __ego_dimensions_length__ that aren't defined in the file, and they're given a value of 0.0. 
However, I'm not certain whether this is directly related to the issue.

Again, I'm not a simulator developer, and since my simulator environment is different, there's a high possibility that my observations lack accuracy.

kyoichi-sugahara, 2024-07-03:
As a premise, while my environment differs in directory structure, the vehicle_model used, and sensor models, I believe we are referring to the same repositories for autoware.universe, autoware_launch, and scenario_simulator_v2.
With that in mind, I performed the following checks in order and was unable to reproduce the issue you reported:

- Run the simulation as described in the tutorial
- Run the simulation using the scenario file provided by @brkay54 san

I confirmed that both ran without any issues :thinking: 

https://github.com/tier4/scenario_simulator_v2/assets/32741405/bc0a4b53-cfe7-4cc5-878a-85b1d5950624

My proposal for now is to update all the repository and build and run simulation.

brkay54, 2024-07-04:
@kyoichi-sugahara -san, thank you for the effort.

> By the way, does the simulator mentioned in the [tutorial](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/scenario-simulation/planning-simulation/scenario-test-simulation/#scenario-test-simulation) work in your environment?

I haven't tried before, but today I tried this scenario. I faced very weird behavior:

https://github.com/tier4/scenario_simulator_v2/assets/45468306/e3bdfff8-1e90-409d-9576-e5b4e43af671

> As a premise, while my environment differs in directory structure, the vehicle_model used, and sensor models, I believe we are referring to the same repositories for autoware.universe, autoware_launch, and scenario_simulator_v2.

My current Autoware workspace is the default files which come with the autoware.repos and simulator.repos. 

> I've only looked at the downloaded files, but I'm a bit concerned that in sample.yaml, there are definitions like ego_dimensions_length that aren't defined in the file, and they're given a value of 0.0.

After this comment, I updated the vehicle dimensions by using the Scenario Editor: 
[image]

But, behavior hasn't changed 😞 .

kyoichi-sugahara, 2024-07-04:
@brkay54 
I'd like to reproduce the situation as it would help debug whether the cause of the issue lies with autoware or with the simulator.
Therefore, I set up autoware from scratch in a separate workspace without using ccache or other tools, and ran the sample scenario. with following command
```
ros2 launch scenario_test_runner scenario_test_runner.launch.py   architecture_type:=awf/universe/20240605   record:=false   scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml'   sensor_model:=sample_sensor_kit   vehicle_model:=sample_vehicle initialize_duration:=90
```
- changed `architecture_type` to awf/universe/20240605  
- changed `initialize_duration` to 90
And log output is following
```
[INFO] [launch]: Default logging verbosity is set to INFO
architecture_type                   := awf/universe/20240605
autoware_launch_file                := planning_simulator.launch.xml
autoware_launch_package             := autoware_launch
consider_acceleration_by_road_slope := False
consider_pose_by_road_slope         := True
enable_perf                         := False
global_frame_rate                   := 30.0
global_real_time_factor             := 1.0
global_timeout                      := 180
initialize_duration                 := 90
launch_autoware                     := True
launch_rviz                         := False
output_directory                    := /tmp
port                                := 5555
record                              := false
rviz_config                         := 
scenario                            := $(find-pkg-share scenario_test_runner)/scenario/sample.yaml
sensor_model                        := sample_sensor_kit
sigterm_timeout                     := 8
use_sim_time                        := False
vehicle_model                       := sample_vehicle
```
[Screencast from 2024年07月05日 01時27分04秒.webm](https://github.com/tier4/scenario_simulator_v2/assets/32741405/4498958a-cb68-4afe-9f8a-71cded7c53f9)

It works also fine :cry:

ahmeddesokyebrahim, 2024-07-04:
Thanks @kyoichi-sugahara -san a lot for your efforts supporting us.
Are you using autoware with source or docker installation ?

ahmeddesokyebrahim, 2024-07-04:
@mitsudome-r -san : I would like to raise your attention here that this problem is representing a blocker for me and @brkay54 to test the changes needed for our tasks.

ahmeddesokyebrahim, 2024-07-04:
Here is the [output](https://youtu.be/-Vt-FaD4VpQ) I see when testing with this [tutorial](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/scenario-simulation/planning-simulation/scenario-test-simulation/#scenario-test-simulation).

brkay54, 2024-07-05:
@kyoichi-sugahara-san, I really appreciate your effort in identifying the problem. 🙏🏼 

After you mentioned ccache, I suspected that the cache on my computer might be causing this issue. I cleared all caches and builds, and then rebuilt the environment, but the problem wasn't resolved.

I tried the exact command you shared:

```bash
berkay@victus:~/projects/autoware$ ros2 launch scenario_test_runner scenario_test_runner.launch.py   architecture_type:=awf/universe/20240605   record:=false   scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml'   sensor_model:=sample_sensor_kit   vehicle_model:=sample_vehicle initialize_duration:=90
[INFO] [launch]: All log files can be found below /home/berkay/.ros/log/2024-07-05-03-12-44-068761-victus-2511435
[INFO] [launch]: Default logging verbosity is set to INFO
architecture_type                   := awf/universe/20240605
autoware_launch_file                := planning_simulator.launch.xml
autoware_launch_package             := autoware_launch
consider_acceleration_by_road_slope := False
consider_pose_by_road_slope         := True
enable_perf                         := False
global_frame_rate                   := 30.0
global_real_time_factor             := 1.0
global_timeout                      := 180
initialize_duration                 := 90
launch_autoware                     := True
launch_rviz                         := False
output_directory                    := /tmp
port                                := 5555
record                              := false
rviz_config                         := 
scenario                            := $(find-pkg-share scenario_test_runner)/scenario/sample.yaml
sensor_model                        := sample_sensor_kit
sigterm_timeout                     := 8
use_sim_time                        := False
vehicle_model                       := sample_vehicle
```

https://github.com/tier4/scenario_simulator_v2/assets/45468306/7cb1df56-557f-4337-9119-ed8d88712bc9

It's quite strange that you don't have this problem, but we do. I'm unsure how to proceed. Tomorrow, I will also kindly ask my colleagues to try it.

kyoichi-sugahara, 2024-07-05:
@ahmeddesokyebrahim 
> Are you using autoware with source or docker installation ?

I'm following source installtion step :thinking: 

I can't debug this because I can't reproduce the problem sorry for this :pray: 
But you guys are having same issue, it's strange...
From my side, I will wait for other people's report to this issue...

kyoichi-sugahara, 2024-07-05:
@maxime-clem 
I heard from other members that you're setting up the development environment for [autoware](https://github.com/autowarefoundation/autoware). Is that true?
If you have the environment at hand, could you please run the following command and check if the phenomenon reported in this issue can be reproduced?
>ros2 launch scenario_test_runner scenario_test_runner.launch.py   architecture_type:=awf/universe/20240605   record:=false   scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml'   sensor_model:=sample_sensor_kit   vehicle_model:=sample_vehicle initialize_duration:=90

maxime-clem, 2024-07-05:
I just built the latest version and I am able to successfully run the scenarios.
Could it be a network configuration issue ? I know the DDS configuration for example can sometimes cause issues.

beyzanurkaya, 2024-07-05:
@brkay54  I built the latest version but I'm facing the same problem. 

https://github.com/tier4/scenario_simulator_v2/assets/32412808/20a69b9e-f6d4-4095-bd94-31be22fd0be7

I ran the same scenario 3-4 times. I also experienced the incident in this video.

>  2024-07-05.03-13-01.mp4 
> It's quite strange that you don't have this problem, but we do. I'm unsure how to proceed. Tomorrow, I will also kindly ask my colleagues to try it.

MertClk, 2024-07-05:
@brkay54 @ahmeddesokyebrahim @beyzanurkaya 
I created an issue to fix the bug. I'll create the PR soon.

https://github.com/tier4/scenario_simulator_v2/issues/1312

ahmeddesokyebrahim, 2024-07-05:
> @brkay54 @ahmeddesokyebrahim @beyzanurkaya I created an issue to fix the bug. I'll create the PR soon.
> 
> #1312

I have tested it for now using multiple scenarios and I can confirm that the mentioned problem is no longer occurring after applying the proposed change.

Thanks a lot @MertClk :clap:

kyoichi-sugahara, 2024-07-09:
I apologize for not being able to assist with analyzing the cause :bow: 
@MertClk 
For future troubleshooting and development reference, I'd like to know: How did you notice this error?
I want to know how to identify the cause when I suspect something might be the reason.

zulfaqar-azmi-t4, 2024-07-19:
@kyoichi-sugahara @MertClk 
I was having similar issues, and confirm that with Mert's solution,the problem is solved.

closed 2024-09-19 by hakuturu583

hakuturu583, 2024-09-19:
Sorry for waiting long time.
This problem was fixed in master branch.

Thank you very much.