tier4/scenario_simulator_v2 #1312: Fix Eigen::Vector3d variable definition

MertClk, 2024-07-05:
https://eigen.tuxfamily.org/dox/TopicPitfalls.html

> In short: do not use the auto keywords with Eigen's expressions, unless you are 100% sure about what you are doing. In particular, do not use the auto keyword as a replacement for a Matrix<> type. 

- https://github.com/tier4/scenario_simulator_v2/issues/1302

```cpp
const Eigen::Vector3d relative_position =
```

This should fix it.

_Originally posted by @MertClk in https://github.com/tier4/scenario_simulator_v2/pull/1226#discussion_r1666832342_

hakuturu583, 2024-07-25:
@MertClk Can you provide information on the operating environment, such as CPU model number/compiler?

We found that this PR implementation changes the behavior of the existing scenario when we check the behavior in the internal regression test environment of TIER IV.

Perhaps the cause of #1312 is a processing or CPU architecture dependent bug due to the Expression Template and the associated computation delay.

Changing the auto type to Eigen::Vector3d seems to change the behavior in some scenarios, and a detailed investigation of the cause and confirmation of the behavior is required to accept this PR.

All computers used in TIER IV are Intel CPUs, gcc compiler and release builds.

MertClk, 2024-07-25:
@hakuturu583 Below you can find my system information.

* Ubuntu 22.04.4 LTS 64-bit
* Intel Core i9-13900HX
* gcc version 11.4.0
* Build Type: RelWithDebInfo

hakuturu583, 2024-07-25:
Thank you very much!
I had thought you were using an AMD CPU.

Thanks for the information.

hakuturu583, 2024-07-25:
@MertClk What eigen version are you using?
Did you install eigen via apt or build eigen by yourself?

MertClk, 2024-07-25:
@hakuturu583 thank you so much for your support on this issue! I'm using eigen version 3.4.0 and installed via apt.

hakuturu583, 2024-07-26:
I use same version... :thinking: 
I am currently looking for the cause of the problem considering multiple possibilities, but we are unable to reproduce the problem on our end and fix it quickly.
I apologize for this.

hakuturu583, 2024-07-29:
It is unclear if it is the model number of the CPU or what is causing the problem,

```bash
const auto relative_position =
````

and

```bash
const Eigen::Vector3d relative_position =
```

There may be a slight difference in calculation results between ``bash const Eigen::Vector3d relative_position = ````.
We will now verify whether there is actually a difference or not by branching out to scenario_simulator_v2.

MertClk, 2024-07-30:
@hakuturu583 
I asked my colleagues who has the same issue and below you can find their CPU models.

* Ryzen 7 6850U 
* Intel i7 12700h
* Intel(R) Core(TM) i9-11900H
* Intel(R) Core(TM) i7-11800H

ahmeddesokyebrahim, 2024-07-30:
I would like to share with you a very important observation.

As a quick reminder, I had this problem re-producible on my machine. The solution from @MertClk is solving it.

I am using Intel(R) Core(TM) i9-11900H and I am always using building with `RelWithDebInfo`.

But I noticed in the reply from @hakuturu583 -san :
> All computers used in TIER IV are Intel CPUs, gcc compiler and **release builds**.

So I have tested scenario simulation (branch `RJD-736/autoware_msgs_support` **without Mert's PR)** with two builds
- with `-DCMAKE_BUILD_TYPE=RelWithDebInfo` 
  -  The issue is still occurring and the vehicle is not moving.
- with `-DCMAKE_BUILD_TYPE=Release`
  - The issue is still NOT occurring and the behavior looks normal.

Initially speaking, it looks like the difference in optimization level turns out to a different generated code to an extent that there is difference in the behavior producing such issue.

I hope this observation would be useful for coming to a final resolution for this issue.

hakuturu583, 2024-07-31:
@ahmeddesokyebrahim Thank you very much!
It is likely that the optimization level is affecting the Expression Template, resulting in incorrect results when calculations are performed with a delay.

The cause can be narrowed down significantly.

hakuturu583, 2024-08-08:
Now, we are investigating why this problem happens.
It is necessary for checking the regression in our regression test environment is a valid behavior or not.
Sorry for taking time.

hakuturu583, 2024-08-26:
Currently, we are writing minimal test code for this problem and verifying its operation with various build options.
The behavior is complex, with calculation results changing depending on factors such as the presence or absence of OpenMP, and we are taking time to analyze the problem.
For now, we would appreciate if you could use scenario_simulator_v2 with a Release build.

closed 2024-09-27 by hakuturu583