tier4/scenario_simulator_v2 #1365: Stuck on "/api/autoware/set/velocity_limit service is not ready" with latest autoware

dtonda8, 2024-09-08:
**To Reproduce**
Steps to reproduce the behavior:
1. Follow the steps on the [Quick Start](https://tier4.github.io/scenario_simulator_v2-docs/user_guide/QuickStart/) up to *Run the examples*
2. Run the following
```bash
ros2 launch scenario_test_runner scenario_test_runner.launch.py global_frame_rate:=10 record:=false scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml' sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle
```

**Expected behavior**
Rviz should pop up and should run simulation

**Logs**
[launch.log](https://github.com/user-attachments/files/16920629/launch.log)

**Desktop (please complete the following information):**
 - Ubuntu Version: 22.04
 - ROS 2 version: Humble
 - No NVidia

hakuturu583, 2024-09-11:
@dtonda8 Thank you for your report.
Could you provide commit hash of the Autoware you used?

dtonda8, 2024-09-11:
Hi @hakuturu583. Here's the commit hash of the Autoware I'm using b17202b090d5adbfcea7c6bf549b9a54da1c8524. ([link to commit](https://github.com/autowarefoundation/autoware/commit/b17202b090d5adbfcea7c6bf549b9a54da1c8524)). I do not have a NVidia graphics card on my laptop (if that matters or not).

hakuturu583, 2024-09-12:
[Screencast from 09-12-2024 02:35:26 PM.webm](https://github.com/user-attachments/assets/7dc0a1d6-fcb1-42f1-ba8f-aa7bd92916bb)

Hmm... It works in my environment...

hakuturu583, 2024-09-12:
Upon checking the logs, it appears that there may be some issues with Autoware in your environment, as topics such as initialpose3d are not being published. This suggests that there might be a problem with the Autoware system.

dtonda8, 2024-09-12:
Yes that seems to be the case. I also needed to reduce the frame rate since my laptop's system is poor (I'm running a 8GB ram VM). I will retry this tomorrow with a proper pc and keep you updated.

dtonda8, 2024-09-13:
Tried with a different pc with fresh Autoware install. It managed to launch RViz but couldn't start the scenario. I think the the issue still lies with the intialpose.

[launch.log](https://github.com/user-attachments/files/16995037/launch.log)

https://github.com/user-attachments/assets/938e6003-513e-4619-85e6-563289206506

dtonda8, 2024-09-16:
Issues with `source install/setup.bash`:
[image]
[image]

Do you think this affects the program?

hakuturu583, 2024-09-18:
It think it is not.
This log means Autoware was build without cuda/tensorrt.

hakuturu583, 2024-09-18:
How about specifying larger values to the launch argument `initialize_duration`. https://github.com/tier4/scenario_simulator_v2/blob/8d189b12b2eca96fbe393bd5f433328647871cb8/test_runner/scenario_test_runner/launch/scenario_test_runner.launch.py#L76C64-L76C83

e.g

```
ros2 launch scenario_test_runner scenario_test_runner.launch.py global_frame_rate:=10 record:=false scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml' sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle initialize_duration:=60
```

dtonda8, 2024-09-22:
It manages to mostly work now! Thanks @hakuturu583 

https://github.com/user-attachments/assets/1709e764-8aa8-43ed-8392-5490cf112ca2

The only issue is that I had to run it a couple of times before it gets working, thanks for sharing advice about the `initialize_duration`.

As you can see on the vid, it's quite laggy, but that's most likely due to my poor system (8 GB Ubuntu VM). If you don't mind me asking, what's the specs of your P.C.?

closed 2024-09-22 by dtonda8