tier4/scenario_simulator_v2 #1374: Error: `random_test_runner_node` throw Unexpected architecture_type "awf/universe"

felixf4xu, 2024-09-12:
**Describe the bug**
I'm trying to follow the instructions from https://autowarefoundation.github.io/autoware-documentation/main/tutorials/scenario-simulation/planning-simulation/random-test-simulation/ to run the scenario simulator:
```sh
ros2 launch random_test_runner random_test.launch.py   architecture_type:=awf/universe   sensor_model:=sample_sensor_kit   vehicle_model:=sample_vehicle
```
I got this error message:

>[random_test_runner_node-1] terminate called after throwing an instance of 'common::scenario_simulator_exception::SemanticError'
[random_test_runner_node-1]   what():  Unexpected architecture_type "awf/universe" given for V2I traffic lights simulation.

I check the code a little bit, it's from

```cpp
#if __has_include(<autoware_perception_msgs/msg/traffic_signal_array.hpp>)
    if (architecture_type == "awf/universe/20230906") {
      return std::make_shared<
        TrafficLightPublisher<autoware_perception_msgs::msg::TrafficSignalArray>>(
        std::forward<decltype(xs)>(xs)...);
      has_made_publisher = true;
    }
#endif
#if __has_include(<autoware_perception_msgs/msg/traffic_light_group_array.hpp>)
    if (architecture_type == "awf/universe/20240605") {
      return std::make_shared<
        TrafficLightPublisher<autoware_perception_msgs::msg::TrafficLightGroupArray>>(
        std::forward<decltype(xs)>(xs)...);
      has_made_publisher = true;
    }
#endif
```

 - Ubuntu Version [e.g. 22.04]
 - ROS 2 version humble

HansRobo, 2024-09-12:
@felixf4xu Thank you for your report!
The latest Autoware-oriented `scenario_simulator_v2` should use `awf/universe/20240605` for `architecture_type` , but this was not reflected in the documentation.
Thank you for noticing.
I'll try to fix it.

HansRobo, 2024-11-19:
I fixed the issue in https://github.com/tier4/scenario_simulator_v2/pull/1375 and https://github.com/autowarefoundation/autoware-documentation/pull/610.
If you are still having issues please reopen or create a new issue

closed 2024-11-19 by HansRobo