tier4/scenario_simulator_v2 #1376: [scenario_editor] More clear error message when "Triangulation of polygon vertices failed"

felixf4xu, 2024-09-12:
This is a problem that I had with https://scenario.ci.tier4.jp/scenario_editor, it does not directly belong to this repo but I hope it's related.

When I uploaded a map made by myself and I had this error:

>Triangulation of polygon vertices failed. This error may occur if there are duplicate vertices.

I think it's useful but it can be more clear if the failed vertices can print their id, so that I can re-edit the map file.

Or even more ideally, https://tools.tier4.jp/vector_map_builder_ll2/ can have this kind of check before exporting the map file.

hakuturu583, 2024-09-27:
Thank you for reporting issue.
I notify to developer of scenario editor.

aratafuji, 2024-10-03:
@felixf4xu 
Sorry for the late reply.
And thank you for your improvement suggestion.
I have added your suggestion to the Scenario Editor's Backlog list.

It will take a while to implement, but we will contact you when it is completed.

aratafuji, 2024-10-28:
@felixf4xu 
> I think it's useful but it can be more clear if the failed vertices can print their id, so that I can re-edit the map file.

The ID of the failed vertex is included in the error message.
[image]

We hope this improvement helps you!
Thanks.

closed 2024-10-29 by felixf4xu