tier4/scenario_simulator_v2 #1383: Can't run scenario simulator, error log files are attached

felixf4xu, 2024-09-16:
**Describe the bug**

I can run autoware planning simulation by `ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/yangzhou-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit`. In the same workspace, I compiled simulator folder and try to run scenario simulator like this:

```
ros2 launch scenario_test_runner scenario_test_runner.launch.py   architecture_type:=awf/universe/20230906   
record:=false   scenario:='$(find-pkg-share scenario_test_runner)/scenario/sample.yaml'   
sensor_model:=sample_sensor_kit   vehicle_model:=sample_vehicle
```

but I got some errors. The rviz splash screen can show but then it quits.

[sceneario-simulator-log.zip](https://github.com/user-attachments/files/17012305/sceneario-simulator-log.zip)

I uploaded the `.ros/log/` folder. It seems that there are 4 log files for each `ros2 launch scenario_test_runner scenario_test_runner.launch.py`.

I checked the log files, one of the `[ERROR]` logs is like this:
```
1726486505.8283932 [ERROR] [component_container_mt-42]: process has died [pid 1185, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args -r __node:=awapi_relay_container 
-r __ns:=/awapi -p use_sim_time:=True -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 
-p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 
-p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7'].
```
it's about `awapi_relay_container`, is that a fatal error?

[autoware-console-log.txt](https://github.com/user-attachments/files/17012309/autoware-console-log.txt)

I also uploaded the console's output.

**Desktop (please complete the following information):**
 - Ubuntu  22.04
 - ROS 2 humble

hakuturu583, 2024-09-18:
The immediate cause is that the simulator terminates the scenario as a failure because Autoware does not transition to WaitingForRoute within initialize_duration seconds.
Try changing the initialize_duration argument in ros2 launch scenario_test_runner scenario_test_runner.launch.py to a larger value, such as initialize_duration:=60.

oguzkaganozt, 2024-09-18:
I am facing the exact same issue even i increased the duration with `initialize_duration:=180` the simulator still waiting for autoware to get into the state of `WaitingForRoute ` for 3 minutes then closes. But i can smoothly run the planning simulation without any problem. Any updates @felixf4xu ? fyi @hakuturu583

hakuturu583, 2024-09-19:
After reading the [launch.log](https://github.com/user-attachments/files/17053023/launch.log), it appears that the following problems may be occurring:

Some nodes have not completed startup and are generating errors.
(Although there are no logs that clearly indicate a crash) Some nodes may have crashed, resulting in necessary topics not being output.

Based on the logs, it seems that very few nodes are being loaded, so the former possibility appears more likely.

hakuturu583, 2024-09-19:
The global_frame_rate does not affect Autoware's initialization, so the effective approach is to set the initialize_duration to a larger value.

mitsudome-r, 2024-09-19:
@felixf4xu Could you try specifying `architecture_type:=awf/universe/20240605` instead of `architecture_type:=awf/universe/20230906` when you launch scenario_test_runner.launch.py?

felixf4xu, 2024-09-19:
increasing value of `initialize_duration` helps, now I can see the rviz is fully displayed and the ego car/planning route in it.

But I still have other issues. Maybe the `scenario/sample.yaml` is too complex as the first scenario example, any simpler project which still use rviz?

I can run `scenario/minimal.yaml` successfully, but it does not use rviz (autoware).

hakuturu583, 2024-09-27:
> I can run scenario/minimal.yaml successfully, but it does not use rviz (autoware).

This scenario does not include Ego (Autoware) entity. So it is not a bug.

felixf4xu, 2024-10-02:
It seems that scenario simulator indeed needs some PC with good CPU performance, in my case, `global_frame_rate:=3` helps to run. It does not guarantees to run successfully, but it increases the  ratio of success.

When it failed to run, I checked the CPU load, it's close to 100% load.

closed 2024-10-02 by felixf4xu