tier4/scenario_simulator_v2 #1402: Compiling error:

felixf4xu, 2024-10-01:
It's strange that I can't compile this code:
```cpp
assert(triggering_scenario_object.is<ScenarioObject>());
```
the full log is like this:
```sh
In file included from /usr/include/c++/11/cassert:44,
                 from /usr/include/nlohmann/detail/macro_scope.hpp:143,
                 from /usr/include/nlohmann/detail/string_escape.hpp:4,
                 from /usr/include/nlohmann/detail/exceptions.hpp:9,
                 from /usr/include/nlohmann/detail/conversions/from_json.hpp:15,
                 from /usr/include/nlohmann/adl_serializer.hpp:6,
                 from /usr/include/nlohmann/json.hpp:60,
                 from /home/ccccc/autoware/src/simulator/scenario_simulator/openscenario/openscenario_interpreter/include/openscenario_interpreter/syntax/value_constraint_group.hpp:18,
                 from /home/ccccc/autoware/src/simulator/scenario_simulator/openscenario/openscenario_interpreter/include/openscenario_interpreter/syntax/parameter_declaration.hpp:21,
                 from /home/ccccc/autoware/src/simulator/scenario_simulator/openscenario/openscenario_interpreter/include/openscenario_interpreter/syntax/parameter_declarations.hpp:19,
                 from /home/ccccc/autoware/src/simulator/scenario_simulator/openscenario/openscenario_interpreter/include/openscenario_interpreter/syntax/misc_object.hpp:23,
                 from /home/ccccc/autoware/src/simulator/scenario_simulator/openscenario/openscenario_interpreter/include/openscenario_interpreter/syntax/entity_object.hpp:18,
                 from /home/ccccc/autoware/src/simulator/scenario_simulator/openscenario/openscenario_interpreter/include/openscenario_interpreter/syntax/scenario_object.hpp:20,
                 from /home/ccccc/autoware/src/simulator/scenario_simulator/openscenario/openscenario_interpreter/src/syntax/relative_clearance_condition.cpp:21:
/home/ccccc/autoware/src/simulator/scenario_simulator/openscenario/openscenario_interpreter/src/syntax/relative_clearance_condition.cpp: In lambda function:
/home/ccccc/autoware/src/simulator/scenario_simulator/openscenario/openscenario_interpreter/src/syntax/relative_clearance_condition.cpp:144:39: error: invalid use of ‘struct openscenario_interpreter::is<openscenario_interpreter::syntax::ScenarioObject>’
  144 |     assert(triggering_scenario_object.is<ScenarioObject>());
      |                                       ^~~~~~~~~~~~~~~~~~
/home/ccccc/autoware/src/simulator/scenario_simulator/openscenario/openscenario_interpreter/src/syntax/relative_clearance_condition.cpp: In lambda function:
/home/ccccc/autoware/src/simulator/scenario_simulator/openscenario/openscenario_interpreter/src/syntax/relative_clearance_condition.cpp:184:53: error: invalid use of ‘struct openscenario_interpreter::is<openscenario_interpreter::syntax::ScenarioObject>’
  184 |                   assert(triggering_scenario_object.is<ScenarioObject>());
      |                                                     ^~~~~~~~~~~~~~~~~~
```

**Desktop (please complete the following information):**
 - Ubuntu 22
 - ROS 2 humble

My gcc version is default:
>gcc --version
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0

yamacir-kit, 2024-10-01:
This is almost certainly due to our implementation mistake.
Until the fix is merged into master, please address this by deleting the line that is causing the error. It is just an assertion, so deleting it is not a problem.
Alternatively, changing the simulator to a release build should also solve the problem; there should be no need to debug build the simulator as well in order to debug build Autoware (unless you want to debug the simulator).

felixf4xu, 2024-10-01:
yes, thanks, I would building `Debug` version, change it to `Release` fixed the issue.

HansRobo, 2024-11-14:
The compilation errors are fixed in #1405.
Thank you for reporting!

closed 2024-11-14 by HansRobo