tier4/scenario_simulator_v2 #1441: Rviz not started when running random test runner and error message

Justin-Xiang, 2024-11-07:
**Describe the bug**
I ran the command `ros2 launch random_test_runner random_test.launch.py`,
however, the Rviz not started, and I got an error message below and then the process ended:
```
[INFO] [launch]: All log files can be found below /home/xxx/.ros/log/2024-11-07-15-29-01-293094-xxx-WS2-1667572
[INFO] [launch]: Default logging verbosity is set to INFO
Value of simulator_type: ['simple_sensor_simulator', 'awsim']
Value of architecture_type: ['awf/universe/20230906', 'awf/universe/20240605']
Test param file ''
Autoware architecture 'awf/universe/20230906'
Vehicle info parameters file supplied: '/home/xxx/Desktop/xxx/autoware/install/sample_vehicle_description/share/sample_vehicle_description/config/vehicle_info.param.yaml'. Parameters passed there override passed via arguments
Simulator model parameters file supplied: '/home/xxx/Desktop/xxx/autoware/install/sample_vehicle_description/share/sample_vehicle_description/config/simulator_model.param.yaml'. Parameters passed there override passed via arguments
[INFO] [random_test_runner_node-1]: process started with pid [1667581]
[INFO] [visualization_node-2]: process started with pid [1667583]
[INFO] [simple_sensor_simulator_node-3]: process started with pid [1667585]
[random_test_runner_node-1] [INFO] [1731022141.395022676] [simulation.random_test_runner_node]: test control parameters: input dir:  output dir: /tmp random test type: 0 test count 5 test_timeout 60
[random_test_runner_node-1] [INFO] [1731022141.396017356] [simulation.random_test_runner_node]: Map path found: /home/xxx/Desktop/xxx/autoware/install/kashiwanoha_map/share/kashiwanoha_map/map
[random_test_runner_node-1] [INFO] [1731022141.409846229] [simulation.random_test_runner_node]: Test suite parameters ego_goal_lanelet_id: -1 ego_goal_s: 0 ego_goal_partial_randomization: false ego_goal_partial_randomization_distance: 25 npcs_count: 10 npc_min_speed: 0.5 npc_max_speed: 3 npc_min_spawn_distance_from_ego: 10 npc_max_spawn_distance_from_ego: 100 name: random_test map_name: kashiwanoha_map
[random_test_runner_node-1] [INFO] [1731022141.409874347] [simulation.random_test_runner_node]: Test case 0 parameters:
[random_test_runner_node-1] [INFO] [1731022141.409878041] [simulation.random_test_runner_node]: Test case 1 parameters:
[random_test_runner_node-1] [INFO] [1731022141.409880560] [simulation.random_test_runner_node]: Test case 2 parameters:
[random_test_runner_node-1] [INFO] [1731022141.409882854] [simulation.random_test_runner_node]: Test case 3 parameters:
[random_test_runner_node-1] [INFO] [1731022141.409885114] [simulation.random_test_runner_node]: Test case 4 parameters:
[random_test_runner_node-1] [INFO] [1731022141.409955057] [simulation.random_test_runner_node]: Generating test 1/5
[random_test_runner_node-1] [INFO] [1731022141.409993764] [simulation.random_test_runner_node]: Goal randomization: full
[random_test_runner_node-1] [INFO] [1731022141.434525985] [simulation.random_test_runner_node]: Generating test 2/5
[random_test_runner_node-1] [INFO] [1731022141.434566812] [simulation.random_test_runner_node]: Goal randomization: full
[random_test_runner_node-1] [INFO] [1731022141.454925924] [simulation.random_test_runner_node]: Generating test 3/5
[random_test_runner_node-1] [INFO] [1731022141.455000134] [simulation.random_test_runner_node]: Goal randomization: full
[random_test_runner_node-1] [INFO] [1731022141.455188041] [simulation.random_test_runner_node]: Goal randomization: full
[random_test_runner_node-1] [INFO] [1731022141.477056665] [simulation.random_test_runner_node]: Generating test 4/5
[random_test_runner_node-1] [INFO] [1731022141.477119345] [simulation.random_test_runner_node]: Goal randomization: full
[random_test_runner_node-1] [INFO] [1731022141.494438997] [simulation.random_test_runner_node]: Generating test 5/5
[random_test_runner_node-1] [INFO] [1731022141.494482167] [simulation.random_test_runner_node]: Goal randomization: full
[random_test_runner_node-1] [INFO] [1731022141.511493915] [simulation.random_test_runner_node]: Saving yaml: /tmp/result.yaml
[random_test_runner_node-1] [INFO] [1731022141.511717261] [simulation.random_test_runner_node]: Running test 1/5
[random_test_runner_node-1] [INFO] [1731022141.511740430] [simulation.random_test_runner_node]: Test description: ego_start_position: lanelet_id: 34426, s: 6.859620412267806, offset: 0, rpy: (x, y, z) : (0, 0, 0) ego_goal_position: lanelet_id: 34507, s: 24.884713983295715, offset: 0, rpy: (x, y, z) : (0, 0, 0) ego_goal_pose: position: (x, y, z), (3769.040000883004, 73799.56158352102, -2.298080283799931), orientation (x, y, z, w), (0, 0, 0.2346604302935948, 0.97207740558786)
[random_test_runner_node-1] npc_descriptions:[0]: name: npc0, start_position: lanelet_id: 34630, s: 0.4311170606721151, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 2.7600657679343756
[random_test_runner_node-1] [1]: name: npc1, start_position: lanelet_id: 34762, s: 10.6110016865125, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 2.417367803168215
[random_test_runner_node-1] [2]: name: npc2, start_position: lanelet_id: 34420, s: 17.972286726916522, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 2.296854265584559
[random_test_runner_node-1] [3]: name: npc3, start_position: lanelet_id: 34564, s: 2.7994700841514124, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 1.1106103177509035
[random_test_runner_node-1] [4]: name: npc4, start_position: lanelet_id: 34441, s: 5.0352954083671975, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 1.8987065067629376
[random_test_runner_node-1] [5]: name: npc5, start_position: lanelet_id: 34762, s: 0.16252458214693308, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 1.7444822575879142
[random_test_runner_node-1] [6]: name: npc6, start_position: lanelet_id: 34705, s: 7.754651931634281, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 1.4399683798422327
[random_test_runner_node-1] [7]: name: npc7, start_position: lanelet_id: 35026, s: 1.970118711235261, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 1.519959188802473
[random_test_runner_node-1] [8]: name: npc8, start_position: lanelet_id: 34636, s: 10.89352478334562, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 1.86001732406248
[random_test_runner_node-1] [9]: name: npc9, start_position: lanelet_id: 34585, s: 13.222348133663877, offset: 0, rpy: (x, y, z) : (0, 0, 0), speed: 1.774565697861986
[random_test_runner_node-1] 
[random_test_runner_node-1] [WARN] [1731022141.531741945] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[simple_sensor_simulator_node-3] [WARN] [1731022141.553754382] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[random_test_runner_node-1] malformed launch argument 'rviz_config:=', expected format '<name>:=<value>'
[random_test_runner_node-1] [ERROR] [1731022146.636070841] [simulation.random_test_runner_node]: common::AutowareError occurred: Autoware process is unintentionally exited. exit code: 1
```

**To Reproduce**
Run the command `ros2 launch random_test_runner random_test.launch.py`

**Expected behavior**
As shown in the document, the rviz will start and npcs will spawn in random locations around the vehicle.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - Ubuntu 22.04
 - ros: humble

piotr-zyskowski-rai, 2025-04-14:
@Justin-Xiang Thank you for reporting the issue! We will take a look at the problem and come back to you shortly 👍

hakuturu583, 2025-05-15:
I assume that this problem has been resolved and Close the Issue.
If you have any problems, please re-open the issue.

closed 2025-05-15 by hakuturu583