tier4/scenario_simulator_v2 #1521: Add vehicle_id Parameter for Launch File Configuration

MertClk, 2025-02-04:
**Description**
Currently, the `scenario_test_runner.launch.py` file does not support a straightforward way to load vehicle-specific configuration files via `vehicle_id,` parameter. This requires users to manually modify launch files or environment variables, which can be error-prone and inconvenient when working with multiple vehicles.

Introducing a `vehicle_id` launch parameter would allow users to specify the vehicle configuration dynamically.

The vehicle_id parameter is essential for launching Autoware in a multi-vehicle repository, as it ensures that each vehicle loads the correct configuration files while preventing conflicts.

**Solution**
- Add `vehicle_id` as a launch parameter to `scenario_test_runner.launch.py` and use it when launching planning_sim internally.

MertClk, 2025-02-04:
What do you think about the necessity of this feature?

hakuturu583, 2025-04-10:
Sorry for our late reply, maintainer team start discussing about it.

MertClk, 2025-04-10:
@hakuturu583 
Thank you! If you decide that this feature could be useful, I can create the PR for the modification.

hakuturu583, 2025-04-14:
The launch file [planning_simulator.launch.xml], which is the launch file called inside scenario_simulator_v2 (https://github.com/autowarefoundation/autoware_ launch/blob/main/autoware_launch/launch/planning_simulator.launch.xml), the vehicle_id cannot be specified except as an environment variable.

Therefore, this Issue is deemed to be difficult to deal with and is therefore Closed.
If necessary, please re-open it.

Translated with DeepL.com (free version)

closed 2025-04-14 by hakuturu583

MertClk, 2025-04-14:
@hakuturu583 
Thank you! I will not re-open this ticket since I think it is not super important and there are workarounds for this issue. However, I want to say that defining vehicle_id is possible as following:
`ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit vehicle_id:=default`