tier4/scenario_simulator_v2 #1573: `scenario_simulator_v2` is not compatible `xmlschema>=4.0.0`

HansRobo, 2025-04-14:
**Describe the bug**

`scenario_simulator_v2` occurs an error with `xmlschema>=4.0.0` 

**To Reproduce**
Steps to reproduce the behavior:

1. ensure `xmlschema` version is greater than or equal to 4.0.0
2. run any scenario
3. see error like below.

```bash
[scenario_test_runner.py-1] Error: failed validating '1' with XsdPatternFacets(['[$][{][ A-Za-z0-9_\\+\\-\\*/%$\\(\\)\\.,]*[\\}]']):
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1] Reason: value doesn't match any pattern of ['[$][{][ A-Za-z0-9_\\+\\-\\*/%$\\(\\)\\.,]*[\\}]']
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1] Schema component:
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1]   <xsd:pattern xmlns:xsd="http://www.w3.org/2001/XMLSchema" value="[$][{][ A-Za-z0-9_\+\-\*/%$\(\)\.,]*[\}]" />
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1] Instance type: <class 'xml.etree.ElementTree.Element'>
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1] Instance:
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1]   <FileHeader description="Scenario where test collision conditions by type" author="abco20" date="2024-10-17T05:57:14.376Z" revMajor="1" revMinor="0" />
[scenario_test_runner.py-1] 
[scenario_test_runner.py-1] Path: /OpenSCENARIO/FileHeader
[scenario_test_runner.py-1] 
```

**Expected behavior**
No errors occur on start-up of scenario execution.すｂ 

**Desktop (please complete the following information):**
 - Ubuntu Version

```bash
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy
```

 - ROS 2 version
 - DDS

```bash
$ env | grep "ROS\|RMW"
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_LOCALHOST_ONLY=1
ROS_DISTRO=humble
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

## How to fix this issue for temporary 

Install `xmlschema` whose version is less than 4.0.0.

For example, 

```bash
pip3 install xmlschema==3.4.5
```

moonseoppkim, 2025-04-25:
Thanks! it's working well!

HansRobo, 2026-10-01:
The CI works well without workaround.
https://github.com/tier4/scenario_simulator_v2/pull/1624

closed 2026-10-01 by HansRobo